espressif/arduino-esp32 #683: the analogRead seems doesn't work

1330154737, 2017-09-29:
`float sensor=25;
float gas_value;
void setup()
{

pinMode(sensor,INPUT);
Serial.begin(115200);

}

void loop()
{

gas_value=analogRead(sensor);
Serial.println(gas_value);
}`
I connected my flame sensor, the results of serial output all show 4095

me-no-dev, 2017-09-29:
try on pin above 32

1330154737, 2017-09-29:
@me-no-dev Thank you for your suggestion, but after the replacement is the same, I just started using this development tool Can you give me a sample program with analogRead

sctv, 2017-09-29:
What board are you using?

1330154737, 2017-09-29:
@sctv i use nodemcu 32s

lbernstone, 2017-09-29:
try using int instead of float.  I'm surprised this doesn't give you an error on compile.

theGamerAut, 2017-10-13:
I got the same problem with the WeMos Wifi Battery Board (Pocket 32 clone) on some pins I get always 4095 and on some I got always 0.

vseven, 2017-10-15:
Try changing your resolution and attenuation:

analogReadResolution(11);        // Default of 12 is not very linear.  Recommended to use 10 or 11 depending on needed resolution.
analogSetAttenuation(ADC_6db);   // Default is 11db which is very noisy.  Recommended to use 2.5 or 6.

Also see this:  https://www.esp32.com/viewtopic.php?f=19&t=2881&start=10#p13739

I've gotten pretty accurate analogReads using a HiLetGo NodeMCU ESP32s and the settings above.

maza70, 2018-03-26:
in my project i had a range from 0-3V and then you need to set the right DB:
example code (i use the ESP32 adc library, Input is GPIO36 aka ADC1_CHANNEL_0 ): 

#include <driver/adc.h>
int readLightSensor(){
  adc1_config_width(ADC_WIDTH_BIT_10);   //Range 0-1023 
  adc1_config_channel_atten(ADC1_CHANNEL_0,ADC_ATTEN_DB_11);  //ADC_ATTEN_DB_11 = 0-3,6V
  return adc1_get_raw( ADC1_CHANNEL_0 ); //Read analog
}

everslick, 2018-04-19:
This issue is closed, because it looks as if it is not a bug or problem with the ESP32 Arduino core or its support libraries. For general API usage questions or help on specific coding challenges, please visit the [arduino-esp32 Gitter channel](https://gitter.im/espressif/arduino-esp32). If you feel this issue was closed in error, reopen it and comment, why you think this is a bug in the Arduino-Core.

closed 2018-04-19 by everslick

thecvely, 2018-10-24:
The esp 32 integrates two 12-bit ACD registers. ADC1 whit 8 channels attached to  GPIOs 32-39 ande ADC2 whit 10 channels in another pins. The thing is the ESP32 uses the ADC2 to manage wifi functions, so if you use Wifi, you can´t use that register. 
See that here https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/adc.html

juliansantos, 2018-10-30:
Here it is the explanation to your problem. 
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/adc.html

umartechboy, 2020-07-11:
ADC2 cannot be used if WiFi is enabled. See https://github.com/espressif/arduino-esp32/issues/440.
And, unfortunately, there is now work around YET.

Make sure your pin is not of ADC2

dev-rijan, 2021-05-17:
I got analogRead(36); always 0