espressif/arduino-esp32 #687: SPIFFS:   if(!file) --> does not work ( Example:   SPIFFS_Test )

kendo55, 2017-09-30:
take the Example:   SPIFFS_Test
---------------------------------------------------------
replace the func setup()  with 

void setup(){
    Serial.begin(115200);
   if(!SPIFFS.begin(true)){
        Serial.println("SPIFFS Mount Failed");
        return;
    }

readFile(SPIFFS, "/never_exist.txt");    // a file that does not exist in SPIFFS
}

void readFile(fs::FS &fs, const char * path){
    Serial.printf("Reading file: %s\n", path);

    File file = fs.open(path);
    if(!file){      //   is never achieved !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

// if ( ! fs.exists(path)){    //  The bypass solution !!!!!!!!!!!!!

        Serial.println("Failed to open file for reading");
        return;
    }
    Serial.print("Read from file: ");
    while(file.available()){
        Serial.write(file.read());
    }
}

### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			20/sept/2017
IDE name:							Arduino IDE1.8.4
Flash Frequency:					80Mhz
Upload Speed:						115200

me-no-dev, 2017-10-01:
yup. this is kinda a limitation in spiffs right now ;) fs.open also returns true for folders in fs and spiffs has no notion of folders so it returns true also. workaround is actually: if(file && !file.isDirectory()) { //exists and its a file }

Gadgit83, 2018-09-02:
I'm not sure if it's the same, but I think related, it would appear that SPIFFS.open() is returning true for files which do not exist at all, which means that .exists() also returns true for files which do not exist.

lbernstone, 2018-09-02:
I don't see that, @Gadgit83.  Can you make a sample file that gives you a false exists?

jestebanes, 2019-05-29:
Same problem here, I found the solution of reading the file size and it is zero if the file doesn't exist.
First if(!configFile) is always true, second is always false, because my files are small and third if(size == 0){ is true if the file doesn't exist and false if the file exists.

Best regards

bool FileSetup::loadSetup(void) {
  File configFile = SPIFFS.open(filename, "r");
  if (!configFile) {
    Serial.printf("Failed to open config file %s\n", filename);
    saveSetup();
    return false;
  }

  size_t size = configFile.size();
  if (size > 2048) {
    Serial.printf("Config file size is too large, %u bytes\n", size);
    saveSetup();
    return false;
  }

if(size == 0){
    Serial.printf("Config file %s empty\n", filename);
    saveSetup();
    return false;
  }

closed 2019-08-14 by stale[bot]