espressif/arduino-esp32 #690: Crash with struct and String class

mitchsf, 2017-10-02:
### Hardware:
Board 						ESP32 DevKitC Revision 1
Core Installation/update date:			9/29/17
IDE name:							Arduino IDE 1.8.4
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
This is a small part of a large program. It compiles, but crashes when run. When settings[80] is reduced to [50] it runs successfully, and when fieldOptions[] is reduced it will run. 

I'm getting rid of the String class and switching to char arrays which is the correct way to do this, but I'm not sure if the error is a bug in the core, that's why I'm bringing it to your attention.

`
### Sketch:
`#include <SPI.h>
#include <NTPClientOne.h>
#include <WiFi.h>
#include <TimeZoneOne.h>
#include <TimeLib.h>
#include <TinyGPS++.h>
#include <WiFiUdp.h>
#include <Preferences.h>
#include <Bounce.h>
#include <ArduinoJson.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>
#include <U8g2lib.h>

struct fields {
  byte type; 
  String fieldPrompt;
  String textDefault; 
  byte numDefault; 
  String heading; 
  String fieldOptions[30]; 
  byte valid[30]; 
  bool returnPrompts; 
  String returnVal; 
};
fields settings[80];
byte numberFields;

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
`

### Debug Messages:
`Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x40078680  PS      : 0x00060b30  A0      : 0x8007879c  A1      : 0x3ffe3cb0  
A2      : 0x40080e28  A3      : 0x3ff0005c  A4      : 0x000008ef  A5      : 0x00040000  
A6      : 0x400d0000  A7      : 0x00000002  A8      : 0x8007867d  A9      : 0x3ffe3c70  
A10     : 0x3ff0002c  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x000008ef  
A14     : 0x00000002  A15     : 0x00000002  SAR     : 0x00000014  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x40078680:0x3ffe3cb0 0x4007879c:0x3ffe3ce0 0x40078a68:0x3ffe3d50 0x400800f9:0x3ffe3e70 0x40007c34:0x3ffe3eb0 0x40000740:0x3ffe3f20

CPU halted.
``
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
No debug info is given.

`Decoding 11 results
0x40080e2c: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 101
0x400800f9: _WindowUnderflow8 at ?? line ?
``

everslick, 2017-10-07:
your code is creating 2720 String objects. I doubt this is what you intend to do.

mitchsf, 2017-10-08:
Yes, that's why I'm moving to char arrays. I just didn't think it should crash. Thanks for the comment. Regards.

closed 2017-10-08 by mitchsf