espressif/arduino-esp32 #692: SPIFFS broken Build

forthlightning, 2017-10-02:
### Hardware:
Board:							Custom board
Running arduino-esp32 as an idf component. Both updated as of this writing

Build fails before it gets to my code: 

```

/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp: In member function 'bool fs::SPIFFSFS::begin(bool, const char*, uint8_t)':
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:33:31: error: 'esp_spiffs_mounted' was not declared in this scope
     if(esp_spiffs_mounted(NULL)){
                               ^
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:38:5: error: 'esp_vfs_spiffs_conf_t' was not declared in this scope
     esp_vfs_spiffs_conf_t conf = {
     ^
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:45:46: error: 'conf' was not declared in this scope
     esp_err_t err = esp_vfs_spiffs_register(&conf);
                                              ^
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:45:50: error: 'esp_vfs_spiffs_register' was not declared in this scope
     esp_err_t err = esp_vfs_spiffs_register(&conf);
                                                  ^
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp: In member function 'void fs::SPIFFSFS::end()':
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:56:31: error: 'esp_spiffs_mounted' was not declared in this scope
     if(esp_spiffs_mounted(NULL)){
                               ^
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:57:55: error: 'esp_vfs_spiffs_unregister' was not declared in this scope
         esp_err_t err = esp_vfs_spiffs_unregister(NULL);
                                                       ^
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp: In member function 'bool fs::SPIFFSFS::format()':
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:68:43: error: 'esp_spiffs_format' was not declared in this scope
     esp_err_t err = esp_spiffs_format(NULL);
                                           ^
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp: In member function 'size_t fs::SPIFFSFS::totalBytes()':
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:79:43: error: 'esp_spiffs_info' was not declared in this scope
     if(esp_spiffs_info(NULL, &total, &used)){
                                           ^
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp: In member function 'size_t fs::SPIFFSFS::usedBytes()':
/.../solsense32/Firmware/solsense_esp32/components/arduino-esp32/libraries/SPIFFS/src//SPIFFS.cpp:88:43: error: 'esp_spiffs_info' was not declared in this scope
     if(esp_spiffs_info(NULL, &total, &used)){
                                           ^
make[1]: *** [libraries/SPIFFS/src//SPIFFS.o] Error 1
make: *** [component-arduino-esp32-build] Error 2
```

me-no-dev, 2017-10-03:
yet something in idf is missing :) that is where all spiffs runs

forthlightning, 2017-10-03:
so should i open there?

me-no-dev, 2017-10-04:
i advise you to delete and download idf again so you are sure that all files are there. something with your update did not work out.

closed 2017-10-27 by forthlightning

harirajrv, 2017-12-28:
I am facing the similar issue. Please tell me how this can be fixed.

forthlightning, 2018-01-04:
deleted and recloned, did the trick