espressif/arduino-esp32 #694: Memory issue since IDF update

luc-github, 2017-10-04:
Hi,
It seems since last idf sync (https://github.com/espressif/arduino-esp32/commit/d27d29719579cd765ef032b76e9a50b71bcf2a84) there is a memory issue
before this commit you can have in a function : a static String variable
like `static String s;` without issue

after this commit if a call is done to the function having a static String then esp crash 

My test code is SPIFFS_test.ino where I just added the static variable declaration, even not used it
```
void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
	static String s;
    Serial.printf("Listing directory: %s\n", dirname);
   ...
```
Here the exception:
the first line is the ESP.getFreeHeap() to check memory
```
199016
Guru Meditation Error of type InstrFetchProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x00000000  PS      : 0x00060730  A0      : 0x800d355a  A1      : 0x3ffcf190  
A2      : 0x3ffc2750  A3      : 0x3f401229  A4      : 0x00000000  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800d345a  A9      : 0x3ffcf170  
A10     : 0x3ffc2750  A11     : 0x00000001  A12     : 0x3ffc9f7c  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x00000000:0x3ffcf190 0x400d3557:0x3ffcf1b0 0x400d0d54:0x3ffcf1e0 0x400d103b:0x3ffcf250 0x400e471f:0x3ffcf280
```
may be it is linked to : https://github.com/espressif/esp-idf/issues/1072#issuecomment-334017229

projectgus, 2017-10-04:
This problem is related to the toolchain, but not the same as the IDF linked above.

Have submitted a PR with a potential fix, please see if it solves the problem for you.

luc-github, 2017-10-04:
Yes it seems solving the issue indeed - thank you so much

me-no-dev, 2017-10-04:
merged,! :)

commit 9d2d1f99 (2017-10-04): Link cxx_guards from IDF libcxx not from libstdc++

Fixes #694
PR #695 (2017-10-04): Link cxx_guards from IDF libcxx not from libstdc++
Fixes #694
commit 3b972e86 (2017-10-04): Link cxx_guards from IDF libcxx not from libstdc++ (#695)

Fixes #694
closed 2017-10-04 by me-no-dev