espressif/arduino-esp32 #696: SPI SLAVE DRIVER

ANKITQ, 2017-10-04:
I WANT SPI SLAVE DRIVER CODE. WHAT I AM DOING IS ,I AM TRANSMITTING A DATA FROM STM32 BOARD AND IT IS TRANSMITTED THROUGH ESP32 WIFI. THE STM32 IS SPI MASTER AND ESP32 IS SOI SLAVE. I need simple spi slave driver to be run on esp32 board. please provide some code snippet 

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Describe your problem here

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

me-no-dev, 2017-10-04:
Why all those capital letters? Here is an IDF example which you can use in Arduino to get started: https://github.com/espressif/esp-idf/blob/master/examples/peripherals/spi_slave/receiver/main/app_main.c

closed 2017-10-04 by me-no-dev

ANKITQ, 2017-10-05:
It is not compiling good for esp32 .the error is Error compiling for board ESP32 Dev module.why is it so ?

me-no-dev, 2017-10-05:
I do not know... what did you compile? where is the debug output (Arduino IDE has debug settings in it's options). You need to give more info if you want help with this.

ANKITQ, 2017-10-05:
The error message is like

Arduino: 1.8.4 (Windows 10), Board: "ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 921600, None"

Build options changed, rebuilding all
C:\Users\Windows\Documents\Arduino\sketch_oct05a\sketch_oct05a.ino: In function 'int main()':

C:\Users\Windows\Documents\Arduino\sketch_oct05a\sketch_oct05a.ino:82:5: sorry, unimplemented: non-trivial designated initializers not supported

     };

     ^

C:\Users\Windows\Documents\Arduino\sketch_oct05a\sketch_oct05a.ino:82:5: sorry, unimplemented: non-trivial designated initializers not supported

C:\Users\Windows\Documents\Arduino\sketch_oct05a\sketch_oct05a.ino:82:5: sorry, unimplemented: non-trivial designated initializers not supported

C:\Users\Windows\Documents\Arduino\sketch_oct05a\sketch_oct05a.ino:89:5: sorry, unimplemented: non-trivial designated initializers not supported

     };

     ^

C:\Users\Windows\Documents\Arduino\sketch_oct05a\sketch_oct05a.ino:89:5: sorry, unimplemented: non-trivial designated initializers not supported

exit status 1
Error compiling for board ESP32 Dev Module.