espressif/arduino-esp32 #698: git update problem

mikekgr, 2017-10-05:
Dear Sirs,
a couple of days, I can not complete the git update for my arduino-esp32 installation in Windows 7.
I got the following error:
Merge branch 'master' of https://github.com/espressif/arduino-esp32.git

# Conflicts:
#	cores/esp32/esp32-hal-i2c.c

Auto-merging cores/esp32/esp32-hal-i2c.c
CONFLICT (content): Merge conflict in cores/esp32/esp32-hal-i2c.c
Automatic merge failed; fix conflicts and then commit the result.

----------------------------- Remove above -----------------------------

### Hardware:
Board:							WEMOS Lolin32
Core update date:			                05/Oct/2017
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:					115200

### Description:
See above!

me-no-dev, 2017-10-05:
Seems that you made a change to a core file and now is not letting you to resync without discarding that change :) 
Try:
```
git fetch origin
git reset --hard origin/master
```

mikekgr, 2017-10-05:
@me-no-dev  although I never touched / changed any core file, yes, your treatment did the trick. 
Thanks a lot Sir.

me-no-dev, 2017-10-05:
great :) maybe something else went wrong along the way.

closed 2017-10-05 by me-no-dev