espressif/arduino-esp32 #700: scanNetworks fails

holgerlembke, 2017-10-05:
Hardware: tested with Heltek/OLED/LORA-Modul and Widora-AIR-Clone (cp210x and ch340 usb chip).

SDK is v3.0-dev-745-gc4e65d6a, all downloaded some days ago.

I use the following simple code:

```
#include <WiFi.h>

const char* ssid = "";
const char* password = "";

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");

  Serial.print("ChipRevision: ");
  Serial.print(ESP.getChipRevision());
  Serial.println();
  Serial.print("SdkVersion: ");
  Serial.print(ESP.getSdkVersion());
  Serial.println();

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  static unsigned long zeit1 = 0;
  if (millis() > zeit1 + 30000) {
    zeit1 = millis();

    Serial.println("1");
    int n = WiFi.scanNetworks();
    Serial.println(n);
    for (int i = 0; i < n; i++)
    {
      Serial.print(WiFi.SSID(i));
      Serial.println();
    }
    Serial.println();

    Serial.println("2");
    n = WiFi.scanNetworks();
    Serial.println(n);
    for (int i = 0; i < n; i++)
    {
      /**/
      Serial.print(WiFi.BSSIDstr(i));
      Serial.print(" ");
      /**/
      Serial.print(WiFi.SSID(i));
      Serial.println();
    }
    Serial.println();
  }
}
```
I get 

[image]

Why is there  difference? What do I do wrong?

holgerlembke, 2017-10-05:
Addendum: checking withing print shows that sometime garbage is printed.

(18)⸮⸮⸮⸮$
⸮Qx⸮⸮⸮⸮⸮⸮⸮(2)
(0)(2)
(14)EasyBox-A41802(2)
(8)SinanNet(2)
(8)Benjamin(2)
(11)Telekom_FON(2)
(11)WLAN-33EQQK(2)
(2)

me-no-dev, 2017-10-09:
This is my result:
```

Booting
ChipRevision: 0
SdkVersion: v3.0-dev-745-gc4e65d6a
Ready
IP address: 192.168.0.2
1
11
your-ssid
MESH_APMODE_023031_000112
nbis-test
HP-Print-EE-ficeto
hristian19
www.networx.bg
rusev71
deni0403
ilhane
www.networx.bg
www.networx.bg

2
9
98:01:A7:E8:C2:2E your-ssid
62:01:94:02:30:31 MESH_APMODE_023031_000112
70:73:CB:BB:04:E7 nbis-test
EC:55:F9:36:9B:EE HP-Print-EE-ficeto
64:66:B3:8F:0B:90 hristian19
66:66:B3:8F:0B:90 www.networx.bg
B0:48:7A:F9:01:7C ilhane
00:27:19:FD:2F:CC rusev71
B2:48:7A:F9:01:7C www.networx.bg

1
10
your-ssid
MESH_APMODE_023031_000112
nbis-test
HP-Print-EE-ficeto
hristian19
www.networx.bg
www.networx.bg
ilhane
veseli601
didkoboy5345

2
9
98:01:A7:E8:C2:2E your-ssid
70:73:CB:BB:04:E7 nbis-test
62:01:94:02:30:31 MESH_APMODE_023031_000112
EC:55:F9:36:9B:EE HP-Print-EE-ficeto
64:66:B3:8F:0B:90 hristian19
66:66:B3:8F:0B:90 www.networx.bg
00:27:19:FD:2F:CC rusev71
B0:48:7A:F9:01:7C ilhane
B2:48:7A:F9:01:7C www.networx.bg

```

me-no-dev, 2017-10-09:
I will close this as issue is most probably in the cables/connection between the computer and the module :)

closed 2017-10-09 by me-no-dev

holgerlembke, 2017-10-09:
No, it is not. Tested with different cables on different usb hubs with different computers now. It is simple there.

me-no-dev, 2017-10-09:
I can not reproduce it though...

holgerlembke, 2017-10-09:
I just got 

```
Booting
ChipRevision: 1
SdkVersion: v3.0-dev-745-gc4e65d6a
Ready
IP address: 192.168.111.147
1a
8
assertion "next > (intptr_t)block" failed: file "/Users/ficeto/Espressif/ESP32/esp-idf/components/heap/./multi_heap.c", line 116, function: get_next_block
abort() was called at PC 0x400d39a7 on core 0

Backtrace: 0x400872f4:0x3ffd82a0 0x400873f3:0x3ffd82c0 0x400d39a7:0x3ffd82e0 0x40086c6f:0x3ffd8310 0x4008703f:0x3ffd8330 0x40082602:0x3ffd8350 0x40082a79:0x3ffd8370 0x4000bec7:0x3ffd8390 0x400f86e2:0x3ffd83b0 0x400f2585:0x3ffd83d0 0x400f41bf:0x3ffd8410 0x401064af:0x3ffd8430

Rebooting...
```

```
Decoding 13 results
0x400d39a7: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c line 63 (discriminator 8)
0x400872f4: invoke_abort at /Users/ficeto/Espressif/ESP32/esp-idf/components/esp32/./panic.c line 553
0x400873f3: abort at /Users/ficeto/Espressif/ESP32/esp-idf/components/esp32/./panic.c line 553
0x400d39a7: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c line 63 (discriminator 8)
0x40086c6f: get_next_block at /Users/ficeto/Espressif/ESP32/esp-idf/components/heap/./multi_heap.c line 116
:  (inlined by) multi_heap_free_impl at /Users/ficeto/Espressif/ESP32/esp-idf/components/heap/./multi_heap.c line 405
0x4008703f: multi_heap_free at /Users/ficeto/Espressif/ESP32/esp-idf/components/heap/./multi_heap_poisoning.c line 202
0x40082602: heap_caps_free at /Users/ficeto/Espressif/ESP32/esp-idf/components/heap/./heap_caps.c line 190
0x40082a79: _free_r at /Users/ficeto/Espressif/ESP32/esp-idf/components/newlib/./syscalls.c line 42
0x400f86e2: free_bss_info at ?? line ?
0x400f2585: wifi_get_ap_list_process at ?? line ?
0x400f41bf: ieee80211_ioctl_process at ?? line ?
0x401064af: ppTask at ?? line ?
```

stickbreaker, 2017-10-09:
Power? 
It could be a hardware fault. These ESP32 are power hungry with WiFi, they can peak over 400mA. If the power supply droops, weird stuff happens.

Try using a powered USB hub that is rated for 1A output.  Most laptops use the 500mA USB 2.

Chuck.

spacehuhn, 2017-11-25:
I can reproduce the error with a program that's quickly writing a lot of data into the RAM. But it only happens with an insufficient power supply (I connected it through a cheap USB hub without any extra power).
My same code runs fine on the same board when plugged into the USB port of my computer directly.

So it might be just a power problem. Hope that helps someone :)