espressif/arduino-esp32 #703: Custom partitions table

NicolaRusso, 2017-10-06:
### Hardware:
Board:							ESP32 Dev Module - WROOM32
Core Installation/update date:			02/jul/2017
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
Hi guys, I'm trying to run a custom partition table on my esp32 board. I've created the .bin file thanks to 
gen_esp32part.py from this .csv file:

# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x1C0000,
app1,     app,  ota_1,   0x1D0000,0x1C0000,
eeprom,   data, 0x99,    0x390000,0x1000,
spiffs,   data, spiffs,  0x391000,0x6F000,

Then I paste the .bin in Arduino\hardware\espressif\esp32\tools\sdk\bin. However when I compile my sketch, I see the same flash partition dimension size (1.3 MB more or less).

Am I missing something?

Thank you!

copercini, 2017-10-06:
When Arduino IDE compiles the sketch it runs `gen_esp32part.exe" -q "esp32/tools/partitions/default.csv" "sketch_name.ino.partitions.bin" `

so editing your default.csv file should take effect =)

copercini, 2017-10-06:
Also, it still will show 1.3MB on IDE independent of flash size due it's hardcoded here: https://github.com/espressif/arduino-esp32/blob/master/boards.txt#L12

NicolaRusso, 2017-10-09:
Thank you for the response! I'll try. Anyway, is there a way to check the current partitions table?

delcomp, 2017-11-30:
Reading the partition table:

size_t ul;
esp_partition_iterator_t _mypartiterator;
const esp_partition_t *_mypart;
ul = spi_flash_get_chip_size(); Serial.print("Flash chip size: "); Serial.println(ul);
  Serial.println("Partiton table:");
  _mypartiterator = esp_partition_find(ESP_PARTITION_TYPE_APP, ESP_PARTITION_SUBTYPE_ANY, NULL);
  if (_mypartiterator) {
    do {
      _mypart = esp_partition_get(_mypartiterator);
      printf("%x - %x - %x - %x - %s - %i\r\n", _mypart->type, _mypart->subtype, _mypart->address, _mypart->size, _mypart->label, _mypart->encrypted);
    } while (_mypartiterator = esp_partition_next(_mypartiterator));  
  }
  esp_partition_iterator_release(_mypartiterator);
  _mypartiterator = esp_partition_find(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_ANY, NULL);
  if (_mypartiterator) {
    do {
      _mypart = esp_partition_get(_mypartiterator);
      printf("%x - %x - %x - %x - %s - %i\r\n", _mypart->type, _mypart->subtype, _mypart->address, _mypart->size, _mypart->label, _mypart->encrypted);
    } while (_mypartiterator = esp_partition_next(_mypartiterator));  
  }
  esp_partition_iterator_release(_mypartiterator);

Making a custom partition table for the Arduino IDE:
1. Make a copy of esp32/tools/partitions/default.csv and name it (my example uvee.csv)
2. Make the required changes
3. Open esp32/boards.txt and find the board you are using
4. Make the following changes (size depends on your configuration)
#esp32doit-devkit-v1.upload.maximum_size=1310720
esp32doit-devkit-v1.upload.maximum_size=1835008
#esp32doit-devkit-v1.build.partitions=default
esp32doit-devkit-v1.build.partitions=uvee

My example uvee.csv file:
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x1C0000,
app1,     app,  ota_1,   0x1D0000,0x1C0000,
eeprom,   data, 0x99,    0x390000,0x1000,
spiffs,   data, spiffs,  0x391000,0x6F000,

For my application I don't need eeprom and spiffs. Eventually I will just remove both partitions to get the max. available size for my app while keeping OTA functionality. Or you can always add a larger serial eeprom to the ESP32 which can handle up to 16MByte of memory.

Note that the Arduino IDE will always write the app0 partition after each compile, but also the partitions at 0x001000 (second-stage bootloader) and 0x008000 (the partition table itself).

beegee-tokyo, 2017-12-17:
@delcomp 
I followed your instructions with exactly the same partition table in uvee.csv.
Then I used your code to read the partition tables.
But the partition table still shows the old sizes. 
@ivankravets @me-no-dev 
I am not on Arduino IDE, but on PlatformIO. Anything different to change partition table on PlatformIO???

delcomp, 2017-12-18:
I only use Arduino IDE. I have no idea on how to get this working on PlatformIO, never used it.

beegee-tokyo, 2017-12-18:
@delcomp, I opened an issue on platformio about this.

copercini, 2018-03-04:
Closed since it's a PlatformIO issue

closed 2018-03-04 by copercini

Abdul-Samad-Saleem, 2018-10-09:
I have tried following the procedure very closely for arduino ide. However my partition table does not get updated.  How can I verify: "" When Arduino IDE compiles the sketch it runs gen_esp32part.exe -q "esp32/tools/partitions/default.csv" "sketch_name.ino.partitions.bin" ""