espressif/arduino-esp32 #705: WiFiSmartConfig save settings (AP, PASS)

yankobogdan, 2017-10-07:
WiFiSmartConfig connection working good. But after reboot changes unsaved. I try to get AP name and PASS to write to EEPROM but both unsuccessful. Any advise?

me-no-dev, 2017-10-08:
can you please post minimal sketch that shows the issue?

bill-orange, 2017-11-05:
I have the same issue. Please see #810.

bill-orange, 2017-12-09:
Has there been any progress in fixing this?  I have a couple of application that just won't work if the settings obtained with smart config are lost.

chinswain, 2017-12-22:
Agree with @bill-orange, can this be standard functionality? I think most users would assume the smart config settings are preserved between reboots, regardless of past experience with the esp8266.

ironoman, 2018-05-30:
Thanks alot @bill-orange . i have been looking for a solution

ehsanmqn, 2018-09-20:
Any updates? Does anyone solved the problem?

bill-orange, 2018-09-20:
Beyond the work-around I posted 9 months ago, I don’t think so.

copercini, 2018-10-21:
There is a solution here: https://github.com/espressif/arduino-esp32/issues/1100#issuecomment-392397250

that saves the SmartConfig credentials to NVS, then load it on boot

closed 2019-08-15 by stale[bot]