espressif/arduino-esp32 #706: pgmspace library string functions causing core panic

stickbreaker, 2017-10-08:
### Hardware:
Board:							WeMos WiFi&Bluetooth Battery
Core Installation/update date:			05/OCT/2017
IDE name:						Arduino IDE 1.8.5
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
pgmspace.h is wrong.

**strncpy(dest,source,0x7fffffff)** is not equivalent to **strcpy(dest,source)**.

strncpy(dest,source,0x7fffffff) correctly copies source to destination, but it then NULL fills then next 2 billion bytes.  (kinda hard on app, to have it's data NULL'd!). 
### Sketch:
```cpp
#include <pgmspace.h>

#define eol "<br>\r\n"
const char ENVPRINT[] PROGMEM={
	"Environment Variables:"eol
	"********************"eol
	"<!--#printenv -->"
	"********************"eol};
	
void setup(){
Serial.begin(115200);
Serial.println("hello");

char buffer[500];
Serial.print(" should work, Arduino AVR coded -> ");
Serial.println(FPSTR(ENVPRINT)); // how Arduino AVR expects
Serial.print("\n should work because ESP32 is not Harvard ->");
Serial.println(ENVPRINT); // how ESP32 can work

strcpy(buffer,ENVPRINT);
Serial.print(" Program Mem copied into buffer, buffer=");
Serial.println(buffer);

Serial.println(" the pgmspace.h defines strcpy_P(dest,src) as strncpy(dest,src,0x7FFFFFFF).\n Which copies the src string to the destination, then continues to overwrite the next 2 billion bytes of memory with NULL");
Serial.println(" This of course causes 'bad things' to happen");

strcpy_P(buffer,ENVPRINT);

Serial.println(buffer);
Serial.print(" Never Arrives!!!");
}

void loop(){}

```

### Debug Messages:
```
************ captured Output
hello
 should work, Arduino AVR coded -> Environment Variables:<br>
********************<br>
<!--#printenv -->********************<br>

 should work because ESP32 is not Harvard ->Environment Variables:<br>
********************<br>
<!--#printenv -->********************<br>

 Program Mem copied into buffer, buffer=Environment Variables:<br>
********************<br>
<!--#printenv -->********************<br>

 the pgmspace.h defines strcpy_P(dest,src) as strncpy(dest,src,0x7FFFFFFF).
 Which copies the src string to the destination, then continues to overwrite the next 2 billion bytes of memory with NULL
 This of course causes 'bad things' to happen
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO,
************ captured Output
```

stickbreaker, 2017-10-09:
I created a fix and tried to create a "Pull" but I received an error message that I don't have permission to create a "Pull".

I don't understand?

Here is my modified pgmspace.h
[pgmspace.zip](https://github.com/espressif/arduino-esp32/files/1367304/pgmspace.zip)

Chuck.

me-no-dev, 2017-10-09:
Merged :)

closed 2017-10-09 by me-no-dev