espressif/arduino-esp32 #709: Request for a new release to pick up ESP-IDF patches

nkolban, 2017-10-09:
The BLE C++ classes have been broken for a while because of some puzzles introduced by the pre3.0 ESP-IDF and new toolchain.  It seems that fixes have been introduced into Git master for ESP-IDF that have resolved those underlying issues.  I'm humbly asking for a new release of the Arduino-ESP32 package that incorporates these changes.

me-no-dev, 2017-10-09:
@nkolban sure, will do as soon as I get a minute :) hopefully today. How soon do you see the BLE part be separate from your cpp utils? Something that we can include in Arduino easier?

nkolban, 2017-10-09:
Howdy my friend.  The BLE libraries are available as an ESP32 architecture Library in the Arduino master library database and can be installed through the Arudino IDE Library manager.  Its been that way for about 3-4 weeks now.

This was our tracking issue:

https://github.com/nkolban/esp32-snippets/issues/56

me-no-dev, 2017-10-09:
Why not add info and how-to in this repo's README? :)

nkolban, 2017-10-09:
I just figured that the BLE C++ classes were "just another library and nothing special" ... don't want to pollute the Arduino ESP32 project with arbitrary references.  I'll be delighted to do what ever you need.  You sir are fearless leader of the Arduino-ESP32 project and you set the direction that we will follow.   If there is something special you want, ask and it shall be done.

SijmenSchoon, 2017-10-13:
Any news on this?

nkolban, 2017-10-13:
Weekend is coming.  That's when I can devote many hours to ESP32 projects.   If we had a new release by then, I could likely test and turn out a new BLE C++ set of classes as a library by Monday.

me-no-dev, 2017-10-13:
I was hoping to see any of this weeks merges come to github and update the libs but that is not the case so far :(

nkolban, 2017-10-13:
Would it be possible to produce a release from ESP-IDF master as of this time without waiting for more/newer changes to come along?  Is there a strategy or process in place for when new Arduino library releases are cut?  Is there something that you are waiting on in the ESP-IDF master that is needed for an Arduino enhancement?

Obviously, you know best.  My goal in asking these is not to push you to do something you don't want to do but rather to understand the process and thinking so that I can set the expectations of users who come to my tiny corner of the world looking for assistance/support.

me-no-dev, 2017-10-13:
The end of the week is usually the best time to push a new build :) At least gives the weekend without any further desync with IDF. Something is broken in the builder on our side and that is why updates are not pushed yet. Will make a new release today (mkspiffs is also in)

reaper7, 2017-10-13:
update branch with new BLE libs 👍 
scan works wery well

me-no-dev, 2017-10-13:
yup :) waiting for travis to finish building

me-no-dev, 2017-10-13:
Update is IN :) BLE as well!

closed 2017-10-13 by me-no-dev