espressif/arduino-esp32 #713: Websocket client support

paulfer747, 2017-10-09:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32
Core Installation/update date:			7OCT17
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
Hi Guys does the ESP32 have any suport for websockets as a websocket client?

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

copercini, 2017-10-12:
@paulfer747 I saw [your post on ESP32 forum](https://esp32.com/viewtopic.php?f=19&t=3269) , what library are you using for make it works?

paulfer747, 2017-10-12:
HI I added it to the post :-)

-- 

Best regards,

Paul Ferreira

Website:         http://www.imagenpaul.com
Cellphone:      0760850990
Email:              paul@imagenpaul.com

logo

---
This email has been checked for viruses by AVG.
http://www.avg.com

paulfer747, 2017-10-12:
And its all thanks to this library!

https://github.com/morrissinger/ESP8266-Websocket

-- 

Best regards,

Paul Ferreira

Website:         http://www.imagenpaul.com
Cellphone:      0760850990
Email:              paul@imagenpaul.com

logo

---
This email has been checked for viruses by AVG.
http://www.avg.com

copercini, 2017-10-13:
@paulfer747 Awesome man!

seaser0, 2018-01-24:
Did you solve your issue?

closed 2018-04-19 by everslick