espressif/arduino-esp32 #714: Wifi UDP client does not work

eohan, 2017-10-10:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			0.10.0
IDE name:							Platform.io
Flash Frequency:					unsure , assuming 80mhz
Upload Speed:						115200

### Description:
I ran this code on the esp32. It connects to the wifi network. I was able to ping the esp32 from the pc. However, the python program https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiUDPClient/udp_server.py does not get any messages from esp. It asked for a permission to access the network, which I agreed 
The pc ip is  "192.168.137.1"

Am I missing something? 

### Sketch:
```cpp

//Change the code below by your sketch
/*
 *  This sketch sends random data over UDP on a ESP32 device
 *
 */
#include <Arduino.h>
#include <WiFi.h>
#include <WiFiUdp.h>

// WiFi network name and password:
const char * networkName = "baloon";
const char * networkPswd = "598468588";

//IP address to send UDP data to:
// either use the ip address of the server or
// a network broadcast address
const char * udpAddress = "192.168.137.1";
const int udpPort = 3333;
IPAddress myIP(192,168,137,12);
IPAddress gateway(192,168,137,1);
IPAddress subnet(255,255,255,0);

//Are we currently connected?
boolean connected = false;

//The udp library class
WiFiUDP udp;

//wifi event handler
void WiFiEvent(WiFiEvent_t event){
    switch(event) {
      case SYSTEM_EVENT_STA_GOT_IP:
          //When connected set
          Serial.print("WiFi connected! IP address: ");
          Serial.println(WiFi.localIP());
          //initializes the UDP state
          //This initializes the transfer buffer
          udp.begin(WiFi.localIP(),udpPort);
          connected = true;
          break;
      case SYSTEM_EVENT_STA_DISCONNECTED:
          Serial.println("WiFi lost connection");
          connected = false;
          break;
    }
}

void connectToWiFi(const char * ssid, const char * pwd){
  Serial.println("Connecting to WiFi network: " + String(ssid));

  // delete old config
  WiFi.disconnect(true);
  //register event handler
  WiFi.onEvent(WiFiEvent);
  WiFi.config(myIP,gateway,subnet);
  //Initiate connection
  WiFi.begin(ssid, pwd);

  Serial.println("Waiting for WIFI connection...");
}

void setup(){
  // Initilize hardware serial:
  Serial.begin(115200);

  //Connect to the WiFi network
  connectToWiFi(networkName, networkPswd);
}

void loop(){
  //only send data when connected
  if(connected){
    Serial.println("printing");
    //Send a packet
    udp.beginPacket(udpAddress,udpPort);
    udp.printf("Seconds since boot: %u", millis()/1000);
    udp.endPacket();
  }
  //Wait for 1 second
  delay(1000);
}

```

### Debug Messages:
```
Using Platform.io. Unable to set debug flags
```

beegee-tokyo, 2017-10-10:
Try to put 
```
  // Initilize hardware serial:
  Serial.begin(115200);
  Serial.setDebugOutput(true);
```
and you can see log message on the serial terminal.

I use UDP to send packages to my PC and it works without problems. But I am not using the example code.
I wrote a subroutine to send the packets:
```
/**
 * udpSendMessage
 * Send a message over UDP
 *
 * @param ipAddr
 *				receiver IP address
 * @param udpMsg
 *				payload to be send
 * @param udpPort
 *				port to be used to send
 * @return <code>bool</bool>
 *				true if payload could be sent
 *				false if error occurs
 **/
bool udpSendMessage(IPAddress ipAddr, String udpMsg, int udpPort) {
	/** WiFiUDP class for creating UDP communication */
	WiFiUDP udpClientServer;

	// Start UDP client for sending packets
	int connOK = udpClientServer.begin(udpPort);

	if (connOK == 0) {
		Serial.println("UDP could not get socket");
		return false;
	}
	udpClientServer.begin(udpPort);
	int beginOK = udpClientServer.beginPacket(ipAddr, udpPort);

	if (beginOK == 0) { // Problem occured!
		udpClientServer.stop();
		Serial.println("UDP connection failed");
		return false;
	}
	int bytesSent = udpClientServer.print(udpMsg);
	if (bytesSent == udpMsg.length()) {
		Serial.println("Sent " + String(bytesSent) + " bytes from " + udpMsg + " which had a length of " + String(udpMsg.length()) + " bytes");
		udpClientServer.endPacket();
		udpClientServer.stop();
		return true;
	} else {
		Serial.println("Failed to send " + udpMsg + ", sent " + String(bytesSent) + " of " + String(udpMsg.length()) + " bytes");
		udpClientServer.endPacket();
		udpClientServer.stop();
		return false;
	}
}
```
Usage like:
```
String udpMsg = "Send package to single address";
IPAddress pcIP (192,	168, 0, 110);
IPAddress multiIP (192,	168, 0, 255);
udpSendMessage(pcIP, udpMsg, 9997);
udpMsg = "BroadCast packet to local network";
udpSendMessage(multiIP, udpMsg, 9997);
```

Works perfect for me.

On the PC (Windows 7) I use [Packet Sender](https://packetsender.com/) to receive packages on the port.

beegee-tokyo, 2017-10-10:
Nope, the python code worked well on my PC.

eohan, 2017-10-20:
Thanks for the help. I realized it was since permissions for network access to both public and private networks were not given to python

closed 2017-10-20 by eohan