espressif/arduino-esp32 #715: error while compiling dmp6 example of I2cdev libary for mpu6050

anhtu0310, 2017-10-10:
Hi , i have been trying to compile the MPU6050_DMP6  example of Jeff Rowberg's i2cdevlib anh i got this 

`n file included from C:\Users\Anh Tu\Documents\Arduino\libraries\MPU6050\examples\MPU6050_DMP6\MPU6050_DMP6.ino:48:0:

C:\Users\Anh Tu\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:62:24: error: conflicting declaration 'typedef int8_t prog_int8_t'

         typedef int8_t prog_int8_t;

                        ^

In file included from C:\Users\Anh Tu\Documents\Arduino\hardware\espressif\esp32\cores\esp32/WString.h:29:0,

                 from C:\Users\Anh Tu\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:149,

                 from C:\arduino_hex\sketch\MPU6050_DMP6.ino.cpp:1:

C:\Users\Anh Tu\Documents\Arduino\hardware\espressif\esp32\cores\esp32/pgmspace.h:25:14: note: previous declaration as 'typedef char prog_int8_t'

 typedef char prog_int8_t;

              ^

In file included from C:\Users\Anh Tu\Documents\Arduino\libraries\MPU6050\examples\MPU6050_DMP6\MPU6050_DMP6.ino:48:0:

C:\Users\Anh Tu\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:66:25: error: conflicting declaration 'typedef int32_t prog_int32_t'

         typedef int32_t prog_int32_t;

                         ^

In file included from C:\Users\Anh Tu\Documents\Arduino\hardware\espressif\esp32\cores\esp32/WString.h:29:0,

                 from C:\Users\Anh Tu\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:149,

                 from C:\arduino_hex\sketch\MPU6050_DMP6.ino.cpp:1:

C:\Users\Anh Tu\Documents\Arduino\hardware\espressif\esp32\cores\esp32/pgmspace.h:29:14: note: previous declaration as 'typedef long int prog_int32_t'

 typedef long prog_int32_t;

              ^

In file included from C:\Users\Anh Tu\Documents\Arduino\libraries\MPU6050\examples\MPU6050_DMP6\MPU6050_DMP6.ino:48:0:

C:\Users\Anh Tu\Documents\Arduino\libraries\MPU6050/MPU6050_6Axis_MotionApps20.h:67:26: error: conflicting declaration 'typedef uint32_t prog_uint32_t'

         typedef uint32_t prog_uint32_t;

                          ^

In file included from C:\Users\Anh Tu\Documents\Arduino\hardware\espressif\esp32\cores\esp32/WString.h:29:0,

                 from C:\Users\Anh Tu\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:149,

                 from C:\arduino_hex\sketch\MPU6050_DMP6.ino.cpp:1:

C:\Users\Anh Tu\Documents\Arduino\hardware\espressif\esp32\cores\esp32/pgmspace.h:30:23: note: previous declaration as 'typedef long unsigned int prog_uint32_t'

 typedef unsigned long prog_uint32_t;`

so can you help me with this issue , thanks

stickbreaker, 2017-10-10:
Comment out one of them. 
You will have to identify which one is unnecessary.

chuck.

agcaahmet, 2017-10-10:
You may also refer to #53

closed 2018-04-19 by everslick