espressif/arduino-esp32 #716: Is system_get_free_heap_size() thread save?

everslick, 2017-10-10:
Why do I ask?

Because I see that ArduinoOTA consumes 97-100% of free heap for no apparent reason for a very brief period of time, probably for only one call to handle(). Even if no update is happening at that time. I know because i have hooked malloc() and friends and check the free heap on each call to malloc(). ESPmDNS and WiFiServer also do the same, BTW. If I disable those libraries these freaky and random yet sparse heap usage spikes are gone. Common to all of them is, that they create a listening socket. The same code on ESP8266 never showed those spikes, so I'm pretty confident my heap metering is OK. My code for heap usage tracking is:

```
#include <lwip/err.h>
#include <stdint.h>

extern void *__real_malloc(uint32_t);
extern void *__real_calloc(uint32_t, uint32_t);
extern void *__real_realloc(void *, uint32_t);
extern void  __real_free(void *);

extern uint32_t mem_free;

static void (*out_of_memory_cb)(void) = NULL;

static void check_pointer(void *ptr) {
  if (!ptr) {
    if (out_of_memory_cb) out_of_memory_cb();
    mem_free = 0;
  }
}

static void check_memory(void) {
  uint32_t heap = system_get_free_heap_size();

  if (heap < mem_free) mem_free = heap;
}

void register_out_of_memory_cb(void (*cb)(void)) {
  // set callback function
  out_of_memory_cb = cb;
}

void *__wrap_malloc(uint32_t size) {
  void *ptr = __real_malloc(size);

  check_pointer(ptr);
  check_memory();

  return (ptr);
}

void *__wrap_calloc(uint32_t num, uint32_t size) {
  void *ptr = __real_calloc(num, size);

  check_pointer(ptr);
  check_memory();

  return (ptr);
}

void *__wrap_realloc(void *ptr, uint32_t size) {
  ptr = __real_realloc(ptr, size);

  check_pointer(ptr);
  check_memory();

  return (ptr);
}

void __wrap_free(void *ptr) {
  __real_free(ptr);

  check_memory();
}
```

The code above only works, when you compile with a custom Makefile, where you are able to alter the linker flags like this: ```LD_FLAGS = -Wl,-wrap,malloc -Wl,-wrap,calloc -Wl,-wrap,realloc -Wl,-wrap,free```

The malloc wrappers will track the free memory in the global uint32_t mem_free, that must be defined elsewhere. I, for one read mem_free every 500ms and draw a graph of the last X readings. After each read, I reset mem_free to 1000000;

[image]

An innocent explanation that I come up with is that code on the other core is messing with the heap during the call to system_get_free_heap_size() and this might result in a wrong/misleading return value.

Another wild guess would be that some dynamic send or receive buffer just fills up rapidly until malloc() returns NULL. either because of a bug, or because network is blocked.

Third option: my code is buggy.

everslick, 2017-10-10:
For clarification I post a log here, explanations below:

```
[01:00:00] ====================================================
[01:00:00]    *** GENESYS V2.0 Firmware V3.0.0  alpha  ***
[01:00:00] ====================================================
[01:00:00] FS:   SPIFFS mounted (total=1342.26KB, used=196.83KB)
[01:00:00] FS:   cannot open file '.240ac405ec54.conf' for reading
[01:00:00] SYS:  calculating delay loop ... 238.57 bogomips
[01:00:00] RTC:  system time (UTC) set to 13:28:50.0
[15:28:50] SYS:  DST is active
[15:28:50] GPIO: activating general purpose IO layer
[15:28:50] LED:  activating status LEDs
[15:28:50] WIFI: scanning for accesspoints ...
[15:28:52] WIFI: found 2 unique SSIDs
[15:28:52] WIFI: using static IP configuration
[15:28:52] WIFI: waiting for STA to connect (THF) ...
[15:28:52] WIFI: SoftAP started with IP '10.1.1.1'
[15:28:52] WIFI: STA connected with local IP '10.0.0.222'
[15:28:52] NTP:  initializing NTP service
[15:28:52] HTTP: initializing webserver
[15:28:52] PROM: initializing AT24C32 eeprom
[15:28:52] WS:   initializing websockets
[15:28:52] TMRY: initializing telemetry (NATIV/MQTT)
[15:28:52] STOR: initializing local storage
[15:28:52] UPD:  initializing OTA update
[15:28:52] MDNS: responder disabled in config
[15:28:52] TLNT: telnet disabled in config
[15:28:52] DHT:  initializing humidity/temperature sensor
[15:28:52] DHT:  no DHT sensor found (checksum error)
[15:28:52] WD:   setting up main loop watchdog
[15:28:52] MAIN: starting main loop ...
[15:28:52] SYS:  CPU is clocked at 240Mhz on ESP32
[15:28:52] SYS:  185644 bytes of total ram, 124220 bytes free
[15:28:52] SYS:  country=DE, language=eng, TZ=+60
[15:28:53] STOR: successfully opened CSV file 'temp.csv'
[15:28:57] TMRY: connected to '10.0.0.150'
[15:29:02] NTP:  sending request to server '81.16.38.161'
[15:29:02] NTP:  system time (UTC) set to 13:29:02.8 (diff=-279ms)
[15:29:03] RTC:  clock set to 9.10.2017 13:29:03
[15:29:25] UPD:  disabling OTA update
[15:39:22] HTTP: 600 seconds of inactivity, forcing logout
[16:28:49] RTC:  system time (UTC) set to 14:28:52.0 (diff=2297ms)
[17:28:51] RTC:  system time (UTC) set to 15:28:54.0 (diff=2092ms)
[18:28:53] RTC:  system time (UTC) set to 16:28:56.0 (diff=2176ms)
[18:29:08] NTP:  sending request to server '91.206.8.36'
[18:29:08] NTP:  system time (UTC) set to 16:29:04.704 (diff=-3932ms)
[18:29:05] RTC:  clock set to 9.10.2017 16:29:05
[19:28:52] RTC:  system time (UTC) set to 17:28:55.0 (diff=2050ms)
[20:28:55] RTC:  system time (UTC) set to 18:28:57.0 (diff=1981ms)
[21:28:56] RTC:  system time (UTC) set to 19:28:59.0 (diff=2009ms)
[21:29:10] NTP:  sending request to server '92.63.212.161'
[21:29:10] NTP:  system time (UTC) set to 19:29:06.782 (diff=-3944ms)
[21:29:07] RTC:  clock set to 9.10.2017 19:29:07
[22:28:54] RTC:  system time (UTC) set to 20:28:57.0 (diff=2037ms)
[23:28:56] RTC:  system time (UTC) set to 21:28:59.0 (diff=2185ms)
[00:28:58] RTC:  system time (UTC) set to 22:29:01.0 (diff=2019ms)
[00:29:12] NTP:  sending request to server '92.63.212.161'
[00:29:13] NTP:  system time (UTC) set to 22:29:09.70 (diff=-3940ms)
[00:29:10] RTC:  clock set to 9.10.2017 22:29:10
[01:28:56] RTC:  system time (UTC) set to 23:28:59.0 (diff=2089ms)
[02:28:58] RTC:  system time (UTC) set to 00:29:01.0 (diff=2185ms)
[03:29:01] RTC:  system time (UTC) set to 01:29:04.0 (diff=2285ms)
[03:29:15] NTP:  sending request to server '83.164.140.130'
[03:29:15] NTP:  system time (UTC) set to 01:29:11.666 (diff=-3948ms)
[03:29:12] RTC:  clock set to 10.10.2017 01:29:12
[04:28:59] RTC:  system time (UTC) set to 02:29:02.0 (diff=2069ms)
[05:29:01] RTC:  system time (UTC) set to 03:29:04.0 (diff=2232ms)
[06:29:03] RTC:  system time (UTC) set to 04:29:06.0 (diff=2185ms)
[06:29:18] NTP:  sending request to server '83.68.137.76'
[... 53 lines cut ...]
[13:54:25] SYS:  WARNING! 99% of heap memory used (1228 bytes free)
[13:55:15] SYS:  WARNING! 99% of heap memory used (1228 bytes free)
[13:55:53] SYS:  WARNING! 99% of heap memory used (1228 bytes free)
[13:55:57] SYS:  WARNING! 99% of heap memory used (1228 bytes free)
[14:01:17] SYS:  WARNING! 98% of heap memory used (3152 bytes free)
[14:06:31] SYS:  WARNING! 99% of heap memory used (1852 bytes free)
[14:18:36] SYS:  WARNING! 98% of heap memory used (2344 bytes free)
[14:29:08] RTC:  system time (UTC) set to 12:29:10.0 (diff=1823ms)
[14:56:09] SYS:  WARNING! 99% of heap memory used (1852 bytes free)
[14:59:39] UPD:  disabling OTA update
[15:09:33] HTTP: 600 seconds of inactivity, forcing logout
[15:29:10] RTC:  system time (UTC) set to 13:29:12.0 (diff=1879ms)
[15:29:24] NTP:  sending request to server '193.170.62.252'
[15:29:24] NTP:  system time (UTC) set to 13:29:20.110 (diff=-3942ms)
[15:29:21] RTC:  clock set to 10.10.2017 13:29:21
[16:29:08] RTC:  system time (UTC) set to 14:29:10.0 (diff=1858ms)
```

[15:29:25] shortly after boot finishes, I manually disable ArduinoOTA.
[08:52:40] I start the telnet server (which creates a listening socket), after that the first spike is logged.
[12:08:35] I activate the ArduinoOTA, many heap usage spikes are cretaed after that.
[12:09:02] I deactivate the telnet server, still spikes are present a lot (from ArduinoOTA).
[14:59:39] I deactivate ArduinoOTA. No spikes since then (3 hours)

Very curious!

everslick, 2017-11-05:
Adding an observation here:

grabbing a big chunk of memory just on startup by mallocing lets say 64KB does not change anything.

the code still runs, but esp_get_free_heap_size() will still report only a few bytes of heap left every few seconds.

everslick, 2019-08-02:
still valid!

atanisoft, 2019-08-02:
Deprecated API... Test with it's replacement or ESP.getFreeHeap() ?

everslick, 2019-08-03:
Thanks, @atanisoft ... I have no explanation what is going on, but after 24 hours I see no spikes in heap usage anymore. I guess the OLDER (IDF 3.1) implementation of `esp_get_free_heap_size()` called `xPortGetFreeHeapSize()` (as you kindly pointed out on gitter) and that might have had a race condition. Replacing `esp_get_free_heap_size()` with `heap_caps_get_free_size()` apparently *fixes* the issue, so I'm going to close this issue.

closed 2019-08-03 by everslick