espressif/arduino-esp32 #717: Wire.begin() causing  unhandled exception

stickbreaker, 2017-10-11:
### Hardware:
Board:							WeMos WiFi&Bluetooth Battery
Core Installation/update date:			09/OCT/2017
IDE name:						Arduino IDE 1.8.5
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
Converting Arduino AVR sketches/ libraries to ESP32. 

If Wire.begin() is called during instantiation  of a library global variable, this Guru Mediation error occurs.  I have not found any discussion outlawing this practice, Arduino AVR will operate correctly.

Is this an error?  

Or is it just my bad programming practices?

Chuck.

### Sketch:  wireFail.h
```cpp
#ifndef wireFail_h
#define wireFail_h

#include <Wire.h>

class wireFail {
public:
  wireFail();
	};
#endif
```
### Sketch:  wireFail.cpp
```cpp
#include <Arduino.h>
#include <wireFail.h>
#include <Wire.h>

wireFail wf;

wireFail::wireFail(){
	Wire.begin();
}
```
### Sketch:  wireFail.ino
```cpp
#include <Arduino.h>
#include <wireFail.h>

void setup(){
Serial.begin(115200);
Serial.print("Never Arrived");
}

void loop(){}
```

### Debug Messages:
```
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400d0a0d  PS      : 0x00060730  A0      : 0x800d0a84  A1      : 0x3ffe3b80  
A2      : 0x3ffc243c  A3      : 0x00000000  A4      : 0x00000016  A5      : 0x000186a0  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d0a09  A9      : 0x3ffe3b60  
A10     : 0x3ffc243c  A11     : 0x00000016  A12     : 0x00000000  A13     : 0x00000064  
A14     : 0x3ffc102c  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x400d0a0d:0x3ffe3b80 0x400d0a81:0x3ffe3ba0 0x400d0a9e:0x3ffe3bc0 0x400d94df:0x3ffe3be0 0x40081239:0x3ffe3c00 0x40081409:0x3ffe3c30 0x4007882e:0x3ffe3c50 0x40078895:0x3ffe3c80 0x400789d1:0x3ffe3cb0 0x40078af7:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:572
load:0x40078000,len:0
load:0x40078000,len:9880
entry 0x400789d8
```

me-no-dev, 2017-10-11:
Wire.begin() is much more complicated on ESP32 than AVR. On AVR you just alter some registers and here you run on OS. Since that class construction happens before the OS is up and running and inside mutexes and other OS things need to be initialized, it's quite "expected" to get and exception.
I guess you can consider this a bad practice on ESP32 :)

closed 2017-10-11 by me-no-dev

stickbreaker, 2017-10-12:
@me-no-dev  Is there a place we can add these type of warning to the documentation?  I propose adding some documentation directly to the Wire library.  Some type of read.me or errata.txt.

Chuck.