espressif/arduino-esp32 #720: sprkfun esp32 things not loading programs

Akskaypohankar, 2017-10-12:
Please fill the info fields, it helps to get you faster support ;)
i am using  Sparkfun esp 32 things board with Arduino all was working well and then it suddenly stopped loading new programmes 
the error is as fallows 

Sketch uses 136570 bytes (10%) of program storage space. Maximum is 1310720 bytes.
Global variables use 11444 bytes (3%) of dynamic memory, leaving 283468 bytes for local variables. Maximum is 294912 bytes.
esptool.py v2.1
Connecting.....Sketch uses 136570 bytes (10%) of program storage space. Maximum is 1310720 bytes.
Global variables use 11444 bytes (3%) of dynamic memory, leaving 283468 bytes for local variables. Maximum is 294912 bytes.
An error occurred while uploading the sketch
esptool.py v2.1
Traceback (most recent call last):
  File "esptool.py", line 2524, in <module>
  File "esptool.py", line 2517, in _main
  File "esptool.py", line 2246, in main
  File "esptool.py", line 177, in __init__
  File "site-packages\serial\__init__.py", line 88, in serial_for_url
  File "site-packages\serial\serialwin32.py", line 62, in open
serial.serialutil.SerialException: could not open port 'COM15': WindowsError(5, 'Access is denied.')
Failed to execute script esptool
the selected serial port Failed to execute script esptool
 does not exist or your board is not connected

Chip is ESP32D0WDQ6 (revision 0)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 8192 bytes to 47..

A fatal error occurred: Timed out waiting for packet content

please help me .

Thank you.
If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Describe your problem here

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

copercini, 2017-10-12:
> serial.serialutil.SerialException: could not open port 'COM15': WindowsError(5, 'Access is denied.')

check if the port is really COM15 and it's not open by another program

copercini, 2017-10-20:
@Akskaypohankar Is it working now?

Akskaypohankar, 2017-10-26:
it was just a piece of solder metal stuck in between pins of ic on the module I cleaned it and it is working normally now still it took 2 days to figure it out :-( 
thanks for the asking @copercini

closed 2017-10-26 by Akskaypohankar