espressif/arduino-esp32 #722: ESP32 HW I2C on alternative GPIO SDA 23 and SCL 19 rather than standard 21/22

blotfi, 2017-10-13:
### Hardware:
Board:							LOLIN32 V1
Core Installation/update date:			13/10/2017?
IDE name:							Arduino IDE 1.8.3
Flash Frequency:					?40Mhz?
Upload Speed:						921600

Hi
I want to use an OLED SSD1306 on SDA 23 and SCL 19 rather than standard 21/22
But I can't use HW
`U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* clock=*/ 19, /* data=*/ 23); ` 

only software emulation I2C work, but of course it is slow comparing to HW I2C
`U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);`
but I know than ESP32 can put the I2C SCL SDA on any GPIO

Any solution to force the I2C to these pins ?

Thanks

Here is the working code with SW I2C
```

#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

#define SDA 23
#define SCL 19

u8g2_uint_t offset1, offset2;     // current offset for the scrolling text
u8g2_uint_t width1, width2;     // pixel width of the scrolling text (must be lesser than 128 unless U8G2_16BIT is defined
const unsigned int text1_y0=31, text2_y0=66;
const char *text1 = "Radio Chaine 3 ";  // scroll this text from right to left
const char *text2 = "Katy Perry - Miss You More ";  // scroll this text from right to left

/*
  U8glib Example Overview:
    Frame Buffer Examples: clearBuffer/sendBuffer. Fast, but may not work with all Arduino boards because of RAM consumption
    Page Buffer Examples: firstPage/nextPage. Less RAM usage, should work with all Arduino boards.
    U8x8 Text Only Example: No RAM usage, direct communication with display controller. No graphics, 8x8 Text only.
*/

// Please UNCOMMENT one of the contructor lines below
// U8g2 Contructor List (Frame Buffer)
// The complete list is available here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp
// Please update the pin numbers according to your setup. Use U8X8_PIN_NONE if the reset pin is not connected

//U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display
//U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA);   // (rotation, [reset [, clock, data]])
//U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* clock=*/ 19, /* data=*/ 23);   // (rotation, [reset [, clock, data]])
U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display

void setup(void) {
  u8g2.begin();
  u8g2.setFont(u8g2_font_logisoso30_tr);  // set the target font to calculate the pixel width
  width1 = u8g2.getUTF8Width(text1);    // calculate the pixel width of the text
  width2 = u8g2.getUTF8Width(text2);    // calculate the pixel width of the text
  u8g2.setFontMode(1);    // enable transparent mode, which is faster
  offset1 = 0;              // start over again
  offset2 = 0;              // start over again
}

void loop(void) {
  u8g2_uint_t x;

	u8g2.setFont(u8g2_font_logisoso30_tr);    // set the target font
   // draw the scrolling text at current offset
  u8g2.clearBuffer();          // clear the internal memory
  x = offset1;
  do {                // repeated drawing of the scrolling text...
    u8g2.drawUTF8(x, text1_y0, text1);      // draw the scolling text
    x += width1;           // add the pixel width of the scrolling text
  } while( x < u8g2.getDisplayWidth() );    // draw again until the complete display is filled
  
  x = offset2;
  do {                // repeated drawing of the scrolling text...
    u8g2.drawUTF8(x, text2_y0, text2);      // draw the scolling text
    x += width2;           // add the pixel width of the scrolling text
  } while( x < u8g2.getDisplayWidth() );    // draw again until the complete display is filled
    

  offset1-=2;              // scroll by one pixel
  if ( (u8g2_uint_t)offset1 < (u8g2_uint_t)-width1 )   offset1 = 0;             // start over again
  offset2-=4;             // scroll by one pixel
  if ( (u8g2_uint_t)offset2 < (u8g2_uint_t)-width2 )   offset2 = 0;             // start over again
    
  //delay(1);             // do some small delay
}
```

CelliesProjects, 2017-10-14:
Wire.begin( I2C_SDA_PIN, I2C_SCL_PIN, 1000000 );

blotfi, 2017-10-14:
not working eather

blotfi, 2017-10-14:
Downloaded last version of Lib, and it is ok, we can provide other oins like that

```
#define SDA 23
#define SCL 19
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, /* clock=*/ SCL, /* data=*/ SDA);   // (rotation, [reset [, clock, data]])

```

closed 2017-10-14 by blotfi