espressif/arduino-esp32 #725: CPU halted - Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU1)

glararan, 2017-10-14:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		Today
IDE name:						Arduino IDE (1.8.5) in Visual Studio 2017
Flash Frequency:					80MHz
Upload Speed:					921600

### Description:
CPU halts if I uncomment xQueueReceive in loop().

Sorry I dont have Exception decode cause I can't get it working. Opened window is empty. Probably I didn't understand tutorial very well? Upload -> Open serial monitor -> open exception window after crash

Using FastLED 3.1.6 and MSGEQ7 1.2.2
https://github.com/FastLED/FastLED
https://github.com/NicoHood/MSGEQ7

### Sketch:
```cpp
#include "FastLED.h"
#include "MSGEQ7.h"

#if CONFIG_FREERTOS_UNICORE
	#define ARDUINO_RUNNING_CORE 0
#else
	#define ARDUINO_RUNNING_CORE 1
#endif

// FastLED
#define LEDS_NUM 5 * 30

// MSGEQ7
#define pinAnalog 34
#define pinReset 5
#define pinStrobe 4
#define MSGEQ7_INTERVAL ReadsPerSecond(50)
#define MSGEQ7_SMOOTH true

CRGB leds[LEDS_NUM];

CMSGEQ7<MSGEQ7_SMOOTH, pinReset, pinStrobe, pinAnalog> MSGEQ7;

QueueHandle_t queue;

int queueSize = 10;

struct qColor
{
	uint8_t red;
	uint8_t green;
	uint8_t blue;

	qColor(uint8_t r, uint8_t g, uint8_t b)
	{
		red = r;
		green = g;
		blue = b;
	}

	qColor()
	{
	}
};

CRGB tmpColor = CRGB(0, 0, 0);

void loop1(void* pvParameters)
{
	MSGEQ7.begin();

	while (1)
	{
		bool newReading = MSGEQ7.read(MSGEQ7_INTERVAL);

		if (newReading)
		{
			printf("%d\t%d\t%d\t%d\t%d\t%d\t%d\n", mapNoise(MSGEQ7.get(0)), mapNoise(MSGEQ7.get(1)), mapNoise(MSGEQ7.get(2)), mapNoise(MSGEQ7.get(3)), mapNoise(MSGEQ7.get(4)), mapNoise(MSGEQ7.get(5)), mapNoise(MSGEQ7.get(6)));

			qColor color = qColor(mapNoise(MSGEQ7.get(0)), mapNoise(MSGEQ7.get(1)), mapNoise(MSGEQ7.get(6)));

			xQueueSend(queue, (void*)&color, (TickType_t)0);
		}
	}

	vTaskDelete(NULL);
}

void setup()
{
	queue = xQueueCreate(queueSize, sizeof(struct CRGB));

	if (queue == NULL)
		printf("Error creating the queue");

	xTaskCreatePinnedToCore(loop1, "loop1", 4096, NULL, 1, NULL, ARDUINO_RUNNING_CORE);

	FastLED.addLeds<UCS1903, 2, GRB>(leds, LEDS_NUM);
}

void loop()
{
	if (queue != NULL)
	{
		qColor newColor;

		if (xQueueReceive(queue, &newColor, (TickType_t)10) == pdTRUE)
		{
			printf("%d\t%d\t%d\n", newColor.red, newColor.green, newColor.blue);
		}
	}

	fill_solid(leds, LEDS_NUM, tmpColor);

	FastLED.show();

	delay(10);
}
```

### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
VMDPE_1:2:3030:0:3030|253	39	253	0	131	227	115
3030_VMDPE
253	39	115
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU1)
Register dump:
PC      : 0x400838ce  PS      : 0x00060034  A0      : 0x80084ee7  A1      : 0x3ffc0590
A2      : 0x3ffc1360  A3      : 0x00060021  A4      : 0x00060c23  A5      : 0x00000020
A6      : 0x00000020  A7      : 0x00060923  A8      : 0xb33f0001  A9      : 0xb33fffff
A10     : 0x00060021  A11     : 0x00000000  A12     : 0x00060021  A13     : 0x00000000
A14     : 0xffffffff  A15     : 0x3ffc4d14  SAR     : 0x00000014  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x400838ce:0x3ffc0590 0x40084ee7:0x3ffc05b0 0x4008387a:0x3ffc05d0 0x400859f0:0x3ffc05f0 0x40081eb5:0x3ffc0600

CPU halted.
```

Thanks for help

copercini, 2017-10-15:
Try add some delay in loop1 ;)

copercini, 2017-10-20:
@glararan it works with delay?

glararan, 2017-10-20:
@copercini  Hi, im home only at weekends so I could test it now.. I added "delay(5)" after if condition in loop1. Halting continues

glararan, 2017-10-22:
I figured out that queues and msgeq7 works. I can read in loop1 data only if FastLED.show() is commented out

glararan, 2017-10-23:
Resolved with https://github.com/FastLED/FastLED/issues/506#issuecomment-338513421

closed 2017-10-23 by glararan