espressif/arduino-esp32 #726: Double entries in SPIFFS directory listing, while files are open.

everslick, 2017-10-14:
When listing a directory on SPIFFS, files can sometimes show up twice, if the files are open for writing at the same time. The two directory entries for identical files always differ in file size. After the directory was listed, the doubled entries vanish and only one entry remains in the list until the file is written the next time.

me-no-dev, 2017-10-18:
Sorry for the late reply. Maybe post this in the original SPIFFS repo? We just call it's API so nothing we can really do :)

closed 2017-10-18 by me-no-dev

everslick, 2017-10-18:
Yes, this could easily be a bug in SPIFFS. Only thing I can say for sure is
that the same code always works on ESP8266 (differences in the Arduino API
aside).

```
// ESP32

bool fs_ls(String &str) {
  File root = fs_open("/");
  File file = root.openNextFile();

  str = "";
  while (file) {
    str += fs_format_name(file.name());
    str += F(" (");
    str += fs_format_size(file.size());
    str += F(")\r\n");

    file = root.openNextFile();
  }

  fs_close(root);

  return (true);
}
```

```
// ESP32

bool fs_ls(String &str) {
  Dir dir = rootfs->openDir("");

  str = "";
  while (dir.next()) {
    str += dir.fileName();
    str += F(" (");
    str += fs_format_size(dir.fileSize());
    str += F(")\r\n");
  }

  return (true);
}
```

If it is a bug in SPIFFS, I guess it was introduced in a later SPIFFS
version, because i think ESP8266 is based on an older version?