espressif/arduino-esp32 #727: Crash at startup when both BLE and WiFi are engaged.

nkolban, 2017-10-14:
When users include both BLE and WiFi in their applications, we get a crash at startup.  There is an issue in the BLE repository which details this in full and can be found here:

https://github.com/nkolban/esp32-snippets/issues/110

Digging into this, we find that the final assertion is caused at `components/esp32/cpu_start.c` in this code section:

```
    portBASE_TYPE res = xTaskCreatePinnedToCore(&main_task, "main",
                                                ESP_TASK_MAIN_STACK, NULL,
                                                ESP_TASK_MAIN_PRIO, NULL, 0);
    assert(res == pdTRUE);
    ESP_LOGI(TAG, "Starting scheduler on PRO CPU.");
    vTaskStartScheduler();
    abort(); /* Only get to here if not enough free heap to start scheduler */
```

specficially, we fail at the `abort()` statement in the last line.  When we look at this code, we see that the stack size for main() is based of `ESP_TASK_MAIN_STACK` which is a synonym for `CONFIG_MAIN_TASK_STACK_SIZE` which is one of our configuration variables.  The problem (as I see it) is that the ESP-IDF code is supplied pre-compiled for us and hence we don't have the opportunity to change this setting.

This issue asks for collaboration on finding a solution.  It may be as simple as increasing the size of `CONFIG_MAIN_TASK_STACK_SIZE` in the build of the ESP-IDF that is distributed with Arduino-ESP32.

SijmenSchoon, 2017-10-16:
Any updates on this? This issue is quite blocking for my project

me-no-dev, 2017-10-16:
the update is that @nkolban did not say what the stack size should be set to in order to work.

SijmenSchoon, 2017-10-19:
We can't tell you that, because it's hidden away from us.

A good first step would be trying to reproduce this problem yourself, and then playing around with the value until it works. Or not, in which case it wasn't the stack size — @nkolban was also just speculating.

SijmenSchoon, 2017-10-28:
I have no idea what fixed it, as I haven't found anything related to in the git log, but it works now! If @nkolban could check if it's fixed for him too, this issue can be closed.

me-no-dev, 2017-10-30:
Thanks for the thumbs down @SijmenSchoon it really made the time I spent into investigating this worth it ;)

closed 2017-10-30 by me-no-dev

SijmenSchoon, 2017-11-04:
After some research, it seems that 55289a45af1e7898ca0b8da037d92a83baaf0083 had fixed it.

<sup>\<offtopic\>
I'm sorry for reacting like I did — I was under a lot of stress, and didn't quite think my reactions through. Thank you for your work on this awesome project.
\</offtopic\></sup>