espressif/arduino-esp32 #729: ESP32 Lolin32 + Arduino IDE 1.8.3, GPIO read not working

blotfi, 2017-10-15:
### Hardware:
Board:							Lolin32
Core Installation/update date:			Last
IDE name:							Arduino IDE 1.8.3
Flash Frequency:					?
Upload Speed:						921600

### Description:
the below program doesn't work, the LED is always off and the input GPIO15 gives zero while it has 3.3V
I also tried with the pin just above GPIO2, same thing

I have an external 10k pullup to 3.3V (measured voltage) and when I press a BP it goes to 0V
The LED is not damaged (it flashes once when connecting USB to power) and if I use the pin GPIO 2, which is not connected to the BP circuit, if I just connect a floating wire than it goes 1 and 0 randomly n but if I put 3.3V then it stays 0V !
I am powering from an USB PC and the 3.3 pins gives 3.3V

strange or I am forgetting something ?
damaged GPIO ?

### Sketch:
```cpp

#include <Arduino.h>

#define KeyEnter 15 
uint8_t led = LOW;
 
//-----------------------------------------------------
void setup(void) {
  Serial.begin(115200);
	pinMode(KeyEnter, INPUT);
	pinMode(BUILTIN_LED, OUTPUT);
 }
//-----------------------------------------------------
void loop(void) {
	led = digitalRead(KeyEnter);
	digitalWrite(BUILTIN_LED, ! led);
  Serial.print("Led = ");
  Serial.println(led);
  delay(100);
}

```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

tonu42, 2017-10-15:
You probably have a wiring issue. Not related to the arduino esp32 code project.

blotfi, 2017-10-15:
yes indeed, I noticed that if I touch the ESP32 pin GPIO15 directly with my finger, than the push button works. (and the program also work)
Is there a problem of ESD or soldering on this boad?

blotfi, 2017-10-16:
ESP32 badly soldered on this Lolin32

closed 2017-10-16 by blotfi