espressif/arduino-esp32 #731: Race condition with semaphores ... manifesting in BLE libraries

nkolban, 2017-10-15:
We have been hunting a BLE library problem for a while which can be read about in great depth here:

https://github.com/nkolban/esp32-snippets/issues/121

Through pure brute force, we narrowed it down to an adverse interaction between some combination of the following:

1) The very latest toolchain with pthreads support
2) The very latest ESP-IDF
3) The very latest Arduino-ESP32
4) Working with semaphores
5) Only some boards

The problem does not appear to manifest in the same logic in ESP-IDF.  See the above defect for all the history and conclusions made so far.  In summary, at this point ... we have found that calling a FreeRTOS Semaphore release without following it with a 10msec delay causes one set of results and a crash while if we introduce a 10msec delay after the Semaphore release, all works.  The errors manifest themselves in internals code rather than in user code.

A circumvention has been added to the BLE libraries to pause for 10msecs after releasing a semaphore.  This isn't ideal but works.  Ideally we can put our heads together and see if we can gain some additional insight into the puzzle.  I am able to recreate on a board in my environment but only on one board (so far).

tonu42, 2017-10-17:
Just to chime in, getting the abort() error on my ESP32 Sparkfun Thing, no issue on ESPea32. Will test on ESP32 lite later.

me-no-dev, 2017-10-17:
How often do you release semaphores? 10ms is ages :)

nkolban, 2017-10-17:
Howdy ... ideally we don't want any delay.  The delay as documented in this issue:

https://github.com/nkolban/esp32-snippets/issues/121

is currently purely to get the user base working until we can resolve the puzzle properly to determine where the race condition is coming from.  There is every likelihood that it is a BLE C++ code issue ... but my goal is to get it working for the community and then we can polish from there.  As found in issue 121, it seems to be a good one occurring on only some boards and not others.

closed 2019-08-15 by stale[bot]