espressif/arduino-esp32 #733: Pthread not supported on BLE write example

mmbraven, 2017-10-16:
### Hardware:
Board:							ESP32 Wemos Lolin32
Core Installation/update date:			Ive just downloaded IDe adn followed isntructions oh setting up libs and toolchain from git
IDE name:					        Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
When i try to run example supplied with SDK BLE_Write i get ESP32 always resting with message E (1279) esp_pthread: pthread_getspecific: not supported!. 

### Debug Messages:
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:812
load:0x40078000,len:0
load:0x40078000,len:10164
entry 0x400789f8
1- Download and install an BLE scanner app in your phone
2- Scan for BLE devices in the app
3- Connect to MyESP32
4- Go to CUSTOM CHARACTERISTIC in CUSTOM SERVICE and write something
5- See the magic =)
E (1279) esp_pthread: pthread_getspecific: not supported!
E (1279) esp_pthread: pthread_setspecific: not supported!
abort() was called at PC 0x400ecc4f on core 0

```

medamarko, 2017-10-16:
This fixed it for me.
https://github.com/nkolban/esp32-snippets/issues/121#issuecomment-336730052

me-no-dev, 2017-10-18:
Arduino has also been updated. Please close this if it is working for you now

closed 2017-10-19 by me-no-dev