espressif/arduino-esp32 #737: WifiServer, socket and debugging lwip

lonerzzz, 2017-10-17:
### Hardware:
Board:							ESP-WROOM-32 custom board
Core Installation/update date:		15/OCT/2017
IDE name:					        Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:					115200

### Description:
Currently I am trying to allocate a WifiServer instance and call the begin() method to get a socket, but in debugging, I can see that the call to socket() in WiFiServer::begin() is returning -1, a memory allocation error according to err.h in lwip directory (if I am referencing the right error source), despite that the heap is 90K.

In trying to debug, I have tried setting various debug settings in the tools/sdk/lwip/lwip/opt.h file but cannot find the parameter for setting the debug level and cannot get any log output. What are the sets of operations required to 1) assign a specific debug level to lwip, 2) get lwip log output to be output 3) anything else to expand upon the -1 that socket() is returning.

I am guessing that as part of the sdk, my changes are having no effect until one of the executables in the tools directory is run but not sure since Arduino detects the file changes.

me-no-dev, 2017-10-17:
you need to run Arduino as IDF component in order to enable that debug (in lwipopts.h)

lonerzzz, 2017-10-17:
Can you explain why there are versions of the file and which to modify? There is
tools/sdk/include/lwip/lwiptopts.h
and
tools/sdk/include/lwip/port/lwiptopts.h

me-no-dev, 2017-10-17:
neither :) components/lwip/... somewhere in IDF

copercini, 2017-10-17:
It's strange, the files `tools/sdk/include/lwip/lwiptopts.h` and `tools/sdk/include/lwip/port/lwiptopts.h` are identical

lonerzzz, 2017-10-17:
Yep, was wondering why they are there. Now more confused if they are not relevant.

me-no-dev, 2017-10-18:
must be an error in my arduino build script. They are there because that is the config that is build with Arduino, so when you use just Arduino, you need this header. If you use as IDF component though, those headers are not included at all and the ones in idf components are. Rebuilding with different config works only in IDF and not in pure Arduino

closed 2019-08-15 by stale[bot]