espressif/arduino-esp32 #739: Vulnerable to KRACK attack (WPA2 attack)?

davejm, 2017-10-17:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			11/jul/2017
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:

It was recently discovered that correctly implemented WPA2 WiFi protocols are vulnerable to attack. I was wondering if the ESP32 is vulnerable and if a patch could be made.

Reference site: https://www.krackattacks.com/

me-no-dev, 2017-10-17:
Patched last night :)

closed 2017-10-17 by me-no-dev

pieman64, 2017-10-17:
@me-no-dev where can we see the patch?

Testato, 2017-10-18:
is there a commit that make evidence of the file changed ?

copercini, 2017-10-18:
Here is more information about this: https://esp32.com/viewtopic.php?f=13&t=3349