espressif/arduino-esp32 #743: How do I send log_e() to Serial()?

stickbreaker, 2017-10-18:
### Hardware:
Board:							WeMos Bluetooth&Battery
Core Installation/update date:			05/OCT/2017
IDE name:						Arduino IDE 1.8.5
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
How do I get log_e("messages") routed to Serial()?

Serial.setDebugOutput(TRUE); 

Does not seem to do anything?

```c++
void setup(){
Serial.begin(115200);
Serial.print(" debug enabled\n");
Serial.setDebugOutput(true);
log_e("important message here");
}
```
### debug Output
```
 debug enabled
```

Chuck.

me-no-dev, 2017-10-18:
Use ESP32 Dev Board and you will see an option to select debug level :)

closed 2017-10-18 by me-no-dev

stickbreaker, 2017-10-19:
@me-no-dev ,  Thanks, Just what I needed.

Chuck.