espressif/arduino-esp32 #744: ETH_LAN8720

Abishek05, 2017-10-18:
### Hardware:
Board:							ESP32 Dev Module- Node MCU
Core Installation/update date:			18/Oct/2017?
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						?921600?

### Description:
Im working with example code ETH_LAN8720, and i get the following error as mentioned below.

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
E (27295) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0x0000
E (28295) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0x0000
E (29295) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0x0000
E (30295) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0x0000
E (31295) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0x0000
E (32295) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0x0000
E (33295) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0x0000
E (34295) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0x0000
E (35295) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0x0000
E (36295) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0x0000
E (37295) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0x0000
E (38295) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0x0000

```

sauttefk, 2017-10-18:
Looks as if there is a problem with the 50MHz reference clock.
- Which LAN8720 board are you using?
- Is the power/clock enable pin (`GPIO17` by default) connected to the crystal oscillator?

Abishek05, 2017-10-18:
[image]

GPIO17 by default is not connected to crystal oscillator, but i have connected externally using jumper wire.

sauttefk, 2017-10-18:
this board has I²C address **1**
`#define ETH_ADDR        1`

Abishek05, 2017-10-18:
I tried, still the same issue. Also tried #define ETH_PHY_ADDR 1 because i found this macro in Eth.h file.

sauttefk, 2017-10-18:
don't use too long wires. these are frequencies not very suitable for a breadbord.
if you have access to a oscilloscope check if you can see the 50MHz in `GPIO0`

pipi61, 2017-10-18:
Try
https://sautter.com/blog/ethernet-on-esp32-using-lan8720/
this work

sauttefk, 2017-10-18:
Well, I know that exactly this little prototype board is working ;-)
But now without the crystal... https://github.com/espressif/esp-idf/pull/1127

me-no-dev, 2017-10-19:
so can I close this now?

sauttefk, 2017-10-19:
@me-no-dev : No, please ask @Abishek05 if his problem is solved. (I assume this is hardware related).
My post was an unrelated, but the mentioned prototype board is mine and I just used it to make a PR to use the ESP32's APPL to generate the reference clock so there is no need for an external crystal oscillator and no more hassle with the clock on `GPIO0` and the bootloader / PHY-power-pin.
Hopefully this PR is merged real soon so I can make the neccessary changes here.

me-no-dev, 2017-10-19:
Thanks @sauttefk :) I saw your PR and looking forward to that merge as well.
@Abishek05 any comments?

James-sjec, 2017-10-22:
I'm using this same board and i don't get any debug messages after changing address to 1 in Arduino IDE. Also i tried using example code from https://github.com/espressif/esp-idf/tree/master/examples/ethernet and i get the error as shown below. Need help!

See the below jpg for eror
[image]

sauttefk, 2017-10-23:
@Abishek05 , @James-sjec
Please send pictures of your cabeling setup.
Please keep in mind that these are 50MHz signals that are not suited for a breadboard.

Abishek05, 2017-10-26:
Thanks @sauttefk . It turned out to be hardware issue like you said. I got it working now.

closed 2017-10-26 by Abishek05

bhcuong2008, 2018-02-06:
Hi,

I've made a custom LAN with LAN8720A. EMAC running with input 50MHz from PHY. And I also get the same error:
E (800480) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0x0000
E (801480) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0x0000

There are clocks on MDC, and MDIO as the attached files. PHY with address 1. There are 2 different pulse kinds. I dont know how to debug next. I think PHY not respond to SMI interface. MDC has clock of 3.06MHz.

[image]
[image]
[image]

Pls help me some hints to debug next. Thank u very much.
Cuong.

bhcuong2008, 2018-02-06:
With image 2, based on SMI read timing, PHY return data for register 0x2 is 0x0007. But esp32 reports error log "Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0x0000"

Actually data is 0x0007 but I dont know the reason why this error.

bhcuong2008, 2018-02-06:
I think that it may be MDC clock is high! This makes SMI interface sampling wrong data! 

How to reduce to 2.5MHz?

sauttefk, 2018-02-06:
Do you have a pullup resistor between 1kΩ and 1.5kΩ on the SMI-MDIO signal?
The signal rises very slowly when no device is driving MDIO.

bhcuong2008, 2018-02-06:
In document, page 254, bits MIICSRCLK in reg EMACMIIADDR_REG 
 https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf

There are only 2 values for settings MDC clock: APB / 42 for clock 80MHz APB or APB/26 for clock 40MHz. But in my case, APB/26 for 80Mhz, so I get ~ 3.07MHz clock on MDC. This violates SMI timing max is 2.5MHz.

So how to debug next?

Yes, I have MDIO 1.5K pull-up.

sauttefk, 2018-02-07:
@bhcuong2008 : I did a few measurements on my working demo setup, so you can compare them with your defective setup.

Correct SMI Frame with LAN8720 if a Ethernet link is established
[image]

SMI Frame **without** LAN8720
[image]

Alternate SMI Frame **without** LAN8720
[image]

bhcuong2008, 2018-02-07:
Thank sauttefk.

I see my 3rd image similar to ur 1st image, just same same. Is Ur MDC clock 2.94MHz?

bhcuong2008, 2018-02-07:
I can change MDC clock in function esp_eth_smi_write, esp_eth_smi_read

Change
REG_WRITE(EMAC_GMACGMIIADDR_REG, 0x1 | ((reg_num & 0x1f) << 6) | ((phy_num & 0x1f) << 11) | (0x3 << 2));

to REG_WRITE(EMAC_GMACGMIIADDR_REG, 0x1 | ((reg_num & 0x1f) << 6) | ((phy_num & 0x1f) << 11) | (0x0 << 2));

Now, MDC clock about 1.9MHz. But new lib is somewhat different with current arduino-esp libs. So error log shows very fast.

sauttefk, 2018-02-07:
I think your signal looks OK.

https://en.wikipedia.org/wiki/Management_Data_Input/Output

state | name | description
-- | -- | --
Z | tristate MDIO | floating MDIO signal (pullup)
PRE_32 | 32-bits of '1' | preamble
ST | 2-bits of '01' | start bits
OP | 2-bits of Opcode | e.g. read or write
PA5 | 5-bits of PHY address | 32 possible PHYs
RA5 | 5-bits of register address | 32 possible registers
TA | 2-bits of turn-around | floating MDIO signal (pullup)
D16 | 16-bits of data | sent by either SME or PHY, depending on OP
Z | tristate MDIO | floating MDIO signal (pullup)

bhcuong2008, 2018-02-07:
This is waveform of reg 0x3 data returned by PHY 8720A. The 1st image is MDC clock ~ 1.9MHz (after I re-compile ethernet lib, then move to arduino-esp ethernet lib).

The 2nd image is response from PHY after read command from EMAC. time is 148ns, it's ok according to SMI specs (0-300ns).

The 3rd image is full read frame and data returned from PHY. The data returned is 0xc0f1. applying mask 0xfff0, the data esp expected is 0xc0f0. It's completely matched. But why esp still reports errors!

[image]
[image]
[image]

bhcuong2008, 2018-02-07:
One note is that I set MDC pin 32, MDIO pin 33.

sauttefk, 2018-02-07:
Could there be a problem with the IOMUX, so that the internal SMI-module is able to write the MDIO-pin but is unable to read it?

bhcuong2008, 2018-02-07:
Exactly there is problem on pin 32, 33 as this 

https://esp32.com/viewtopic.php?f=12&t=1408&p=6508#p6458

I'm trying to make it work, but still not.

sauttefk, 2018-02-07:
My changes should have fixed your problem: https://github.com/espressif/esp-idf/commit/157371effb5813110c303322025205df02b8282c
Are you sure you are using a recent idf library?
Or is anything **after** it was initialised by the ethernet library.

bhcuong2008, 2018-02-07:
I see no difference vs my esp-idf in func phy_rmii_smi_configure_pins(uint8_t mdc_gpio, uint8_t mdio_gpio):

    gpio_matrix_out(mdc_gpio, EMAC_MDC_O_IDX, 0, 0);
    PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[mdc_gpio], PIN_FUNC_GPIO);
    gpio_matrix_out(mdio_gpio, EMAC_MDO_O_IDX, 0, 0);
    gpio_matrix_in(mdio_gpio, EMAC_MDI_I_IDX, 0);
    PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[mdio_gpio], PIN_FUNC_GPIO);

I use ethernet example of arduino-esp32:
https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples/ETH_LAN8720

The code is just a little. I dont know whether arduino lib has made any changes to IOMUX.

sauttefk, 2018-02-07:
Did you define all pins and stuff according to your setup?
```
#define ETH_CLK_MODE    ETH_CLOCK_GPIO0_IN
#define ETH_POWER_PIN   ????
#define ETH_TYPE        ETH_PHY_LAN8720
#define ETH_ADDR        0 or 1 ?
#define ETH_MDC_PIN     32
#define ETH_MDIO_PIN    33
```

bhcuong2008, 2018-02-07:
Hi Sauttefk,

It's just worked! I modify the code as:
    gpio_set_direction(mdio_gpio, GPIO_MODE_INPUT);

    gpio_matrix_out(mdc_gpio, EMAC_MDC_O_IDX, 0, 0);
    PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[mdc_gpio], PIN_FUNC_GPIO);
    gpio_matrix_out(mdio_gpio, EMAC_MDO_O_IDX, 0, 0);
    gpio_matrix_in(mdio_gpio, EMAC_MDI_I_IDX, 0);
    PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[mdio_gpio], PIN_FUNC_GPIO);

Note, gpio_set_direction must before gpio_matrix_in(mdio_gpio, EMAC_MDI_I_IDX, 0);

If let it after, it does not work.

sauttefk, 2018-02-07:
Great ! This would have been my next suggestion, while I was digging through the docs.

Would you please also try this?
`gpio_set_direction(mdio_gpio, GPIO_MODE_INPUT_OUTPUT)`
Or this?
`gpio_set_direction(mdio_gpio, GPIO_MODE_INPUT_OUTPUT_OD)`

sauttefk, 2018-02-07:
So I think, we should create a pull request...

bhcuong2008, 2018-02-07:
You should do it :) I feel very happy for my 1st design with LAN.

bhcuong2008, 2018-02-07:
There are some updates that I face this issue.

1. The 1st is errata in english document as above (vs Chinese version) (page 254, bits MIICSRCLK in reg EMACMIIADDR_REG)

2. MDC clock should be less than 2.5MHz, not 3.09MHz.

3. IO for MDIO/MDC. 

Thank u very much for ur guide.

bhcuong2008, 2018-02-07:
I already tried this: 
gpio_set_direction(mdio_gpio, GPIO_MODE_INPUT_OUTPUT_OD) 

right after:
gpio_matrix_in(mdio_gpio, EMAC_MDI_I_IDX, 0);

But not success. At that time, MDIO pin always zero. Not output data to PHY.

danielcampillo, 2018-02-07:
Hi, in the example shown above, I changed 
config.phy_addr = CONFIG_PHY_ADDRESS;
to
config.phy_addr = 0;

It has something to do with a I2c address which has the value of 1.

It works now

Saludos desde Envigado/Colombia

sauttefk, 2018-02-07:
@danielcampillo : you can change PHY-address in the setup using `make menuconfig` ---> Example Configuration  ---> PHY Address (0-31)
This is not I²C but SMI (Serial Management Interface)

sauttefk, 2018-02-07:
I modified phy_common.c so that it configures the SMI pins correctly
```
void phy_rmii_smi_configure_pins(uint8_t mdc_gpio, uint8_t mdio_gpio)
{
    // setup SMI MDC pin
    gpio_set_direction(mdc_gpio, GPIO_MODE_OUTPUT);
    gpio_matrix_out(mdc_gpio, EMAC_MDC_O_IDX, 0, 0);
    PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[mdc_gpio], PIN_FUNC_GPIO);
    // setup SMI MDIO pin
    gpio_set_direction(mdio_gpio, GPIO_MODE_INPUT_OUTPUT);
    gpio_matrix_out(mdio_gpio, EMAC_MDO_O_IDX, 0, 0);
    gpio_matrix_in(mdio_gpio, EMAC_MDI_I_IDX, 0);
    PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[mdio_gpio], PIN_FUNC_GPIO);
}
```
Pull request follows soon...

bhcuong2008, 2018-02-08:
Hi sauttefk,

Do we need to check whether it is normal pin or special pin (32-33). Because only pin 32-33 has this issue. Also, it should be GPIO_MODE_INPUT_OUTPUT_OD (if used instead of GPIO_MODE_INPUT) to avoid conflict.

I give some test cases that I did as reference later. 

1. If let 

gpio_set_direction(mdio_gpio, GPIO_MODE_INPUT);

after 

gpio_matrix_out(mdio_gpio, EMAC_MDO_O_IDX, 0, 0);
gpio_matrix_in(mdio_gpio, EMAC_MDI_I_IDX, 0);

Then pin MDIO becomes input forever. Value returned always 0xffff

2. If let 

gpio_set_direction(mdio_gpio, GPIO_MODE_INPUT_OUTPUT_OD);

after 

gpio_matrix_out(mdio_gpio, EMAC_MDO_O_IDX, 0, 0);
gpio_matrix_in(mdio_gpio, EMAC_MDI_I_IDX, 0);

Then MDIO becomes ouput low forever. Value always 0x0000

sauttefk, 2018-02-08:
> Do we need to check whether it is normal pin or special pin (32-33). Because only pin 32-33 has this issue. 

No, we should initialise any given GPIO correctly and should not rely on the bootloader or hardware init values.

> Also, it should be GPIO_MODE_INPUT_OUTPUT_OD (if used instead of GPIO_MODE_INPUT) to avoid conflict.

No, defenitely not. During the output phaseof the master, the pin has to be in a push/pull configuration (I've checked this on the scope - using OpenDrain gives you much to slow slew rates when to signal return to high)

> I give some test cases that I did as reference later.

Yes that were also my findings. I had similar issues when I was creating my APLL patch for the EMAC-CLK

bhcuong2008, 2018-02-08:
About GPIO_MODE_INPUT_OUTPUT_OD, we should confirm to specs. And let users choose suitable rise time, help EMI better. If high rise time, it means that more EMI problems.

About conflict, I means that in some cases (dont know in advance), ESP drives it high, and PHY drives it low. There will be high current flows through this path. And one of them will be damaged. And maybe ESP MDIO pin may be damaged 1st due to low sink/source current, 6mA.

So pls be serious about this.

sauttefk, 2018-02-08:
> About GPIO_MODE_INPUT_OUTPUT_OD, we should confirm to specs.

We do! The MDIO state machine does its job correctly when set to  `GPIO_MODE_INPUT_OUTPUT` (and **not** when set to `GPIO_MODE_INPUT_OUTPUT_OD`).

MDIO is floating before a transfer, is driven high during the preamble, push-pulled in the command and write phase, and floating from the beginning of the TA (turnaround)-phase during a read.

_You can see those different rising edges (push/pull [cyan&purple] and open-drain [red]) on this scope picture (you referred them as "2 different pulse kinds")_
[image]

bhcuong2008, 2018-02-08:
I understood that inside ESP32 will handle when input, output. I mean specs is SMI specs. MDIO is bi-directional, and multi-drop (multi-PHY), like I2C. Everything is not perfect 100%. This practice will help protect our circuit in long-term. 

With my setup, 1.5K pull-up as my above images, rise time is about less than 50ns. This is ok. If u try to reach 1ns rise time, circuit still functions properly. But may lead to EMI issues. Why we want to do so?

So It depends on u to choose suitable approachs.

sauttefk, 2018-02-08:
I think you don't have to worry about EMI on MDIO when there is MDC with 3MHz push/pulled by the master.
Also the negative edge still has the same fast slew rate.

bhcuong2008, 2018-02-08:
To falling-edge, it still has about < 50ns, in my images. You can scope ur waveform larger.

EMI related mainly to rise/falling time, not by frequency.

zekageri, 2018-09-03:
Hello guys!
I have a problem with esp-Wroom-32 with ETH_Lan8720 board. I'am using an arduino example code from here :

https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/ETH_LAN8720/ETH_LAN8720.ino

I did the wireing like this: 

GPIO17 - PHY_POWER   : NC - Osc. Enable - 4k7 Pulldown
GPIO22 - EMAC_TXD1   : TX1
GPIO19 - EMAC_TXD0   : TX0
GPIO21 - EMAC_TX_EN  : TX_EN
GPIO26 - EMAC_RXD1   : RX1
GPIO25 - EMAC_RXD0   : RX0
GPIO27 - EMAC_RX_DV  : CRS
GPIO00 - EMAC_TX_CLK : nINT/REFCLK (50MHz) - 4k7 Pullup
GPIO23 - SMI_MDC     : MDC
GPIO18 - SMI_MDIO    : MDIO
GND                  : GND
3V3                  : VCC

from the site :
https://sautter.com/blog/ethernet-on-esp32-using-lan8720/

I soldered the NC PIN on the lan8720 board to the enable pin of the oscillator. I added pull down resistor to nc pin and pull up to gpio00.

LAN 8720 board : https://fr.aliexpress.com/item/Smart-Electronics-LAN8720-module-network-module-Ethernet-transceiver-RMII-interface-development-board-for-arduino/32825173408.html?spm=a2g0s.9042311.0.0.40696c370I7nRH

ESP32: https://fr.aliexpress.com/item/ESP-WROOM-32-WiFi-Bluetooth-4-2-dual-core-CPU-MCU-low-power-Bluetooth-240MHZ-ESP32/32829367382.html?spm=a2g0s.9042311.0.0.15636c37H4la82

I get the following errors:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
E (2256) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0x0000
E (3256) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0x0000
E (4256) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0x0000

I dont know what to do , can anyone help me?

X-WL, 2018-11-16:
> Hello guys!
> I have a problem with esp-Wroom-32 with ETH_Lan8720 board. I'am using an arduino example code from here : [...]

Hi I use MH-ET Live board. I have the same problem. What is the solution? I am using the latest version of the library in the PlatformIO.

HankLloydRight, 2023-10-27:
Hello. I know this is an old thread, but I'm hoping someone can help me with the LAN8720.

I have a custom built PCB project that is an 8 channel DAC that decodes WAV files from an SD card.

I want to add ethernet to the device. I have this unit: Smart Electronics LAN8720 Module  
I followed all the directions and posts in this thread and here:
https://sautter.com/blog/ethernet-on-esp32-using-lan8720/ (big thanks to sautterfk).

I wired everything up and I have the 8720 working like a dream now with just wires soldered to my existing PCB.  Very stable once I shortened all the wires to about 1"-2" each. 

The problem I now have is that _only_ when I initialize the EMAC, I lose all connection to my SD card.  If I run the ESP32 in Wifi mode or no network mode, the SD card works fine (as it always has). 

Here are the details for both. 
| GPIO | RMII Signal
| -- | -- 
| 0 | REF_CLK 
|21 | TX_EN 
|19 | TX0 
|22 | TX1 
|25 | RX0 
|26 | RX1 
|27 | CRS_DV 
|17 | PHY_POWER (from sautter's page)
|12 | MDC
|13 | MDIO

SD_MMC pinout (running in 1-line mode)
| GPIO | SD_MMC Signal
|-- | --
|14 | CLK
|15 | CMD
|2 | DAT0

Here's my code:
```
#define ETH_MDC_PIN 12
#define ETH_MDIO_PIN 13
listDir(SD_MMC, "/",0);  // this properly lists the contents of the SD card
ETH.begin( 1 ,17, ETH_MDC_PIN, ETH_MDIO_PIN , ETH_PHY_LAN8720, ETH_CLK_MODE);
listDir(SD_MMC, "/",0);  // this returns two errors below
```
The errors I get are:
E (1700989) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x109
E (1700990) diskio_sdmmc: sdmmc_read_blocks failed (265)

I did come across this note here: https://github.com/espressif/esp-idf/tree/master/examples/ethernet

> If you observe undefined behavior (e.g. LCD glitches) of any SPI device which works normally when Ethernet is not connected over internal EMAC, you need to adjust EMAC DMA burst length (the DMA is shared resource between EMAC and the SPI). The same applies when you observe Ethernet frames corruption at the output of SPI Ethernet module and you use combination of internal EMAC and SPI Ethernet module as network interfaces. To configure the EMAC DMA burst length, modify internal Ethernet initialization as follows:
esp32_emac_config.dma_burst_len = ETH_DMA_BURST_LEN_4; // or other appropriate value

But I'm using Arduino Framework on Platform.io, so I'm not sure how to adjust the DMA Burst Length (if that is even the issue). 

Any help, comments, or suggestions why this might be happening would be most helpful. Thank you.

me-no-dev, 2023-10-27:
@HankLloydRight this setting will be available in 3.0.0 (because only exists in newer ESP-IDF) and the current location where this will need to be set is: https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/src/ETH.cpp#L96 (code will change before release). You can test it now with installing the 3.0.0-alpha2 release through the dev board manager link.

HankLloydRight, 2023-10-27:
> @HankLloydRight this setting will be available in 3.0.0 (because only exists in newer ESP-IDF) and the current location where this will need to be set is: [...]

It's funny-- that's the first thing I tried, but obviously did not work:
[image]

I will try to test with 3.0.0-alpha2 today.  Thank you.

HankLloydRight, 2023-10-30:
@me-no-dev -- I can't seem to load the 3.0.0-alpha2 tag into Platform.io -- see my previous post. I don't see any way on the "dev board manager link"
 Do you have any pointers on how to get access to this branch so I can test the new dma_burst_len variable? Thank you.

me-no-dev, 2023-10-30:
You need Arduino IDE or install from git. Maybe @Jason2866 can help you to get PIO going.

HankLloydRight, 2023-10-30:
Thank you. There's no way I could backport my code into Arduino IDE.  I hope @Jason2866 has some suggestions for PIO.

Jason2866, 2023-10-31:
@HankLloydRight You can use my Platformio espressif32 fork.
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.10.12/platform-espressif32.zip
```
It is based on latest IDF 5.1 and Arduino master commit. SPIFFS, Rainmaker, I2S and SR and mbedtls is removed, since we don't use or need.

HankLloydRight, 2023-10-31:
Thank you.

  I did a build with your platform and there seem to be a lot of changes to the timer code? 

Like timerBegin() now takes just a frequency (in Hz?) instead of a timer handle and prescaler? And the other function names seemed to have changed, i.e. timerAlarmWrite() to timerWrite() and the parameters are different.  And the ETH.Begin() has a different parameter order.

Many warnings against using  var++; or var--; for volatile variables (but easily fixable) 

And once I do get to the point of starting ETH.Begin(), I get this error that halts the device entirely:

`[  2909][D][esp32-hal-spi.c:181] spiDetachBus(): Stopping SPI BUS`

And since I also couldn't get a timer started with the new timer functions (I tried several different iterations), I'll have to pass on this update for now and put my ETH PHY+SD_MMC project on hold.

Jason2866, 2023-10-31:
Arduino 3.0.0 has some breaking changes. Timer API is one of them.

HankLloydRight, 2023-10-31:
Thanks again.  Are these changed documented?

HankLloydRight, 2023-10-31:
Ok, thanks. I found the migration guide, I'll give it another go.

HankLloydRight, 2023-11-01:
Hi @Jason2866 and @me-no-dev -- 
Ok, I've loaded the 3.0.0-alpha2 framework (thank you) and now have the updated Timer API working, but now I have two other issues. 

 One, the ETH initializes correctly, but I can never get a DHCP address(see below). There are no router MAC blocks or firewalls. This same ETH PHY board did connect and get an IP address using the previous framework. 

Code:
```
#define ETH_POWER_PIN 17
// Type of the Ethernet PHY (LAN8720 or TLK110)
#define ETH_TYPE ETH_PHY_LAN8720
// I2C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
#define ETH_ADDR 1
// Pin# of the I2C clock signal for the Ethernet PHY
#define ETH_MDC_PIN 12
// Pin# of the I2C IO signal for the Ethernet PHY
#define ETH_MDIO_PIN 13
#define ETH_CLK_MODE ETH_CLOCK_GPIO0_IN
ETH.begin( ETH_PHY_LAN8720, ETH_ADDR , ETH_MDC_PIN, ETH_MDIO_PIN ,ETH_POWER_PIN, ETH_CLK_MODE);
ETH.printInfo(Serial);
```
Output:
```
[  4506][V][WiFiGeneric.cpp:439] _arduino_event_cb(): Ethernet Started
[  4507][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 18 - ETH_START
[  4508][V][WiFiGeneric.cpp:432] _arduino_event_cb(): Ethernet Link Up
[  4515][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 20 - ETH_CONNECTED
eth0: <UP,100M,FULL_DUPLEX,AUTO>
      ether 24:4C:AB:2E:42:4B phy 0x1
      inet 0.0.0.0 netmask 0.0.0.0 broadcast 255.255.255.255
      gateway 0.0.0.0 dns 0.0.0.0

ETH.localIP(): 0.0.0.0
```

And a related problem -- as soon as Ethernet is enabled, I lose access to my SD_MMC SD card on HSPI
` SD_MMC.begin("/sdcard", true, false, SDMMC_FREQ_DEFAULT);`

Accessing the SD card doesn't produce any warnings or errors, it just returns no access when opening the FS.

Thoughts?

Jason2866, 2023-11-01:
@HankLloydRight The network part will not be as it is now in release of core 3.0.0
It has known issues.... You can get inspirations from the working implementation for the current version from here https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_xdrv_driver/xdrv_82_esp32_ethernet.ino

[7 more replies]