espressif/arduino-esp32 #747: Slow compile time under Arduino IDE

ikorman, 2017-10-20:
Hi

I have installed and configured Arduino for ESP32 and from functional point of view it compiles and flashes the boards. Problem is that compile for empty sketch takes 1-2 minutes. From what I can see in the compile log is that cause for that is that each time complete core is compiled from scratch. Is there a way to confgure IDE to do only incremental compile, meaning only source files which have been changed (like for example in PlatformIO environment)?

Arduino IDE v1.8.4
arduino-esp32 (17th of October)

Thx.
Ivan

me-no-dev, 2017-10-20:
IDE compiles everything only the first time, given that the sketch is saved and settings unchanged, but even then it is still slow because arduino-builder goes through all files and included libs to check for changes. It really sucks but that is how it works :( maybe submit an issue to the arduino-builder repository...

closed 2017-10-20 by me-no-dev

copercini, 2017-10-20:
Take a look here: https://github.com/espressif/arduino-esp32/issues/724

The arduino.cc guys are working to make the builder faster, but it's still under heavy work...

ikorman, 2017-10-20:
Thx. I was under impression that each time he recompiles core from scratch as this is what I can see in the log each time I build:

"C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/app_trace" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/heap" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/jsmn" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls_port" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/soc" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/sdmmc" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/spiffs" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/wear_levelling" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/console" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/coap" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/expat" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/json" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/nghttp" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/lwip" -std=gnu99 -Os -g3 -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wpointer-arith -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -MMD -c -DF_CPU=240000000L -DARDUINO=10804 -DARDUINO_NodeMCU-32S -DARDUINO_ARCH_ESP32  -DESP32 -DCORE_DEBUG_LEVEL=0 "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32\variants\nodemcu-32s" "C:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-sigmadelta.c" -o "C:\Users\no060\AppData\Local\Temp\arduino_build_1705\core\esp32-hal-sigmadelta.c.o"

copercini, 2017-10-20:
It just compiles "from zero" at the first time when open your IDE, keeping it open the next sketches will use cached files speeding up the process

ikorman, 2017-10-20:
Hm, in my case does not look to be working like that. Here is what i tried just now:

I started Arduino IDE, compiled the first empty sketch, then started new sketch in new window (leaving previous one open in background), started to build it and again same time and entries like above in compiler log.

me-no-dev, 2017-10-20:
Is your sketch saved?

ikorman, 2017-10-20:
Ups, no :( Sorry for that ..

Repeated the test and indeed it's faster, but sill several time slower that classic Arduino.

Thx. will explore alternative scenarios, like PlatformIO or Visual Studio until this is brought to acceptable compile time.

danielfaust, 2020-06-22:
> Take a look here: #724
> 
> The arduino.cc guys are working to make the builder faster, but it's still under heavy work...

Great to see that they haven't fixed this. It's 2020 now by the way.

The best thing about this is when it issues the message

> Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch

because even then everything will get recompiled... Worst IDE ever.

DebugOk, 2021-03-02:
> Great to see that they haven't fixed this. It's 2020 now by the way.
>
> The best thing about this is when it issues the message
>
>
> because even then everything will get recompiled... Worst IDE ever.

2021 and still waiting...

OzmoOzmo, 2021-03-13:
I gave up on the Arduino IDE as its far too slow for ESP32.
I found Visual Studio Micro to be really fast - compiles and uploads in seconds - but you need to disable the option for "Deep Library Search" 
- which is defaulted to be On and causes two minute compiles if left on.
As a bonus you get debugging too! Yay for breakpoints!

whitevast, 2024-09-06:
2024... Half-Life 3 похоже выйдет быстрее...

cxemnet, 2025-05-01:
2025 still waiting...