espressif/arduino-esp32 #748: esp32 dac adc function

chenkaiyao, 2017-10-20:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		11/jul/2017
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:					115200

### Description:
1.  the pin(25) is a dac pin.  I call dacWrite(25, 0), the pin(25) output voltage should be 0v, but output is 0.1v. 
2. the pin(35) is a abc pin. put 0.1v to the pin(35), I call analogRead(35), the values measured should be 124, but the value is 0.

Is this normal?

### Sketch:
```cpp
void setup()
{
    Serial.begin(115200);
    pinMode(35, INPUT);
}

void loop()
{
    dacWrite(25, 0);
    int a = analogRead(35);
    Serial.printf("a = %d\r\n", a);
    delay(2000);
}
```

### Debug Messages:

Khizer-Jan, 2017-10-20:
Following code is used to read adc value.
`#include <driver/adc.h>

...

    adc1_config_width(ADC_WIDTH_12Bit);
    adc1_config_channel_atten(ADC1_CHANNEL_0,ADC_ATTEN_0db);
    int val = adc1_get_voltage(ADC1_CHANNEL_0);
`
you have to set adc_width (9-12)
for pin 35 adc channel is ADC1_CHANNEL_7)
The input voltage of ADC will be reduced by setting ADC_ATTEN.

Please refer to  [link.](https://esp-idf.readthedocs.io/en/v2.0/api/peripherals/adc.html)

chenkaiyao, 2017-10-21:
@Khizer-Jan adc default value is ADC_WIDTH_12Bit and ADC_ATTEN_11db.
when input voltage change, the pin 35 measured value is change. but 0-0.1v can't measured, the value return 0.

me-no-dev, 2017-10-21:
@Khizer-Jan you refer to IDF API and we have our own

me-no-dev, 2017-10-21:
@chenkaiyao yes there are some “limitations” that are looked at. BTW not sure about the DAC but will have it looked at

closed 2019-08-15 by stale[bot]