espressif/arduino-esp32 #749: Assert reboot on wifi begin in dual core mode with arduino running as a component

forthlightning, 2017-10-20:
hitting this assert

` esp-idf/components/freertos/./tasks.c:1066 (prvAddNewTaskToReadyList)- assert failed! `

I see this code in arduino WiFiGeneric.cpp
```
static void _start_network_event_task(){
    if(!_network_event_queue){
        _network_event_queue = xQueueCreate(32, sizeof(system_event_t *));
        if(!_network_event_queue){
            log_e("Network Event Queue Create Failed!");
            return;
        }
    }
    if(!_network_event_task_handle){
        xTaskCreatePinnedToCore(_network_event_task, "network_event", 4096, NULL, 2, &_network_event_task_handle, 1);
        if(!_network_event_task_handle){
            log_e("Network Event Task Start Failed!");
            return;
        }
    }
    esp_event_loop_init(&_network_event_cb, NULL);
}
```

tries to start task on core 1, no core 1 available because single core mode

saw this assert in tasks.c

```

	/* Assure that xCoreID is valid or we'll have an out-of-bounds on pxCurrentTCB
	   You will assert here if e.g. you only have one CPU enabled in menuconfig and
	   are trying to start a task on core 1. */
	configASSERT( xCoreID == tskNO_AFFINITY || xCoreID < portNUM_PROCESSORS);
```

forthlightning, 2017-10-20:
switched on dual core mode, looks like it fixed the assert reboot.

me-no-dev, 2017-10-21:
Oops :) this could be changed to ```ARDUINO_RUNNING_CORE``` instead

closed 2017-10-21 by me-no-dev