espressif/arduino-esp32 #752: Can't compile anything under Arduino IDE

albydnc, 2017-10-21:
### Hardware:
Board:							DOIT esp32
Core Installation/update date:			Last one (@21/10/17)
IDE name:							Arduino 1.8.3
Flash Frequency:					Default
Upload Speed:						Default

### Description:
Can't compile any sketch for esp32 on Windows 10 x64, arduino ide x86.

### Debug Messages:
Arduino:1.8.3 (Windows 10), Scheda:"DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Alberto Perro\AppData\Local\Arduino15\packages -hardware C:\Users\Alberto Perro\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Alberto Perro\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Alberto Perro\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=921600,DebugLevel=none -ide-version=10803 -build-path C:\Users\ALBERT~1\AppData\Local\Temp\arduino_build_538932 -warnings=default -build-cache C:\Users\ALBERT~1\AppData\Local\Temp\arduino_cache_159045 -prefs=build.warn_data_percentage=75 -verbose C:\Users\ALBERT~1\AppData\Local\Temp\arduino_modified_sketch_110289\GetChipID.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Alberto Perro\AppData\Local\Arduino15\packages -hardware C:\Users\Alberto Perro\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Alberto Perro\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Alberto Perro\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=921600,DebugLevel=none -ide-version=10803 -build-path C:\Users\ALBERT~1\AppData\Local\Temp\arduino_build_538932 -warnings=default -build-cache C:\Users\ALBERT~1\AppData\Local\Temp\arduino_cache_159045 -prefs=build.warn_data_percentage=75 -verbose C:\Users\ALBERT~1\AppData\Local\Temp\arduino_modified_sketch_110289\GetChipID.ino
Using board 'esp32doit-devkit-v1' from platform in folder: C:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32
Using core 'esp32' from platform in folder: C:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32
Detecting libraries used...
"C:\Users\Alberto Perro\AppData\Local\Arduino15\packages\Simba\tools\xtensa-esp32-elf-gcc\1.22.0-59/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_trace" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/heap" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/jsmn" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls_port" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/soc" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/sdmmc" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spiffs" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wear_levelling" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/console" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/coap" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/expat" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/json" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nghttp" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/lwip" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti  -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10803 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32  -DESP32 -DCORE_DEBUG_LEVEL=0 "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32\cores\esp32" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32\variants\doitESP32devkitV1" "C:\Users\ALBERT~1\AppData\Local\Temp\arduino_build_538932\sketch\GetChipID.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\Alberto Perro\AppData\Local\Arduino15\packages\Simba\tools\xtensa-esp32-elf-gcc\1.22.0-59/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_trace" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IC:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Use [... cut ...]
C:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/soc/soc/soc.h:167:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'

             ASSERT_IF_DPORT_REG((_r), REG_SET_BIT);                                                                    \

             ^

C:\Users\Alberto Perro\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/soc/soc/io_mux_reg.h:111:5: note: in expansion of macro 'REG_SET_BIT'

     REG_SET_BIT(PIN_NAME, FUN_PD);

     ^

exit status 1
Errore durante la compilazione per la scheda DOIT ESP32 DEVKIT V1.

Monarch73, 2017-10-23:
Currently esp32 build environment needs to be setup manually in Arduino IDE. What instructions did you follow to set it up? Are you sure you haven't missed out a step?

albydnc, 2017-10-23:
Yes, I followed the instructions on the github page. I installed it also on my Mac and there it is working just fine

Monarch73, 2017-10-23:
Somehow I can't imagine this to be a general problem with windows. Have you tried to pull the latest changes from the repository and try to compile again? 

If that won't work, I'd recommend to delete the repository folder and redo the installation from scratch.

albydnc, 2017-11-06:
I tried to pull the latest changes and recompile it, same error. Reinstalled from scratch, same error.

me-no-dev, 2017-11-06:
what windows is this? I have never seen this error

albydnc, 2017-11-06:
Windows 10 x64 version 1703 on a Dell XPS 15 9550

albydnc, 2017-12-02:
Still Not Working, please help.

stickbreaker, 2017-12-02:
based on the 'strike through' display of your original message, you have a tilde character in your Window's user name. The tilde character is being interpreted as a command parameter by the compiler. you are going to have to change your window's user name or create a new user that does not have a tilde character in it.

Chuck.

everslick, 2017-12-02:
@stickbreaker WOW! good catch. :+1: (if this is indeed the problem, but anyway, incredible observation) :1st_place_medal:  :-D

Monarch73, 2017-12-02:
Sorry, but no.  The tilde can't be the issue. On Windows Systems the tilde is used to indicate that this is a old fashion (dos) 8.3-Filename. It often resolved by Windows functions to get the canonical Path. If the tilde was to cause these problems, we  would see this error way more frequent.

everslick, 2017-12-02:
But still it could be related. Sometimes we see 'C:\Users\Alberto Perro\...' With the full user name and a little bit further down we have the VFAT name 'C:\Users\ALBERT~1\...' that triggers the MARKDOWN strike through. Maybe a mix of different filesystems is at play here?

stickbreaker, 2017-12-02:
@everslick ,Perhaps @Monarch73 is correct, the tilde may not be the cause.  But it is a symptom.  How are parameters detected by the compiler / linker? Aren't they detected / separated by spaces?  The full windows user name of the Op is `C:\Users\Alberto Perro` when this command line is pass down stream, I bet `C:\Users\Alberto` is executed, and fails.

This Issue will still resolve to down to a username/tilde problem.

Chuck.

albydnc, 2017-12-02:
I have changed my user name to a shorter one with no spacing but nothing has changed, still can't compile.  Same error.

stickbreaker, 2017-12-02:
@albydnc look at the directories passed to the compiler, have the directory names changed, or is has Window's just changed your displayed name?   I think Windows is Lazy an will just change your displayed name.  Changing the directory structure would require changing all references in the registry, and all program local configuration files. I think Windows' will just display a new name, and not mess with the directory structure.

Use file manager to look at the directory `C:\Users\`.  Is your base directory still `C:\Users\Alberto Perro\`?

As a test, create a new user with a simple name, install Arduino 1.8.5, and Arduino-esp32 in that new user.

Chuck.

albydnc, 2017-12-02:
@stickbreaker I renamed the directory to c:\Users\Alberto\ then re installed arduino and esp32. nothing changed. also in a new user environment, same error.

albydnc, 2017-12-02:
`Arduino:1.8.5 (Windows 10), Scheda:"DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Alberto\AppData\Local\Arduino15\packages -hardware C:\Users\Alberto\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Alberto\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Alberto\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=921600,DebugLevel=none -ide-version=10805 -build-path C:\Users\Alberto\AppData\Local\Temp\arduino_build_404925 -warnings=default -build-cache C:\Users\Alberto\AppData\Local\Temp\arduino_cache_722578 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32\examples\ChipID\GetChipID\GetChipID.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Alberto\AppData\Local\Arduino15\packages -hardware C:\Users\Alberto\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Alberto\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Alberto\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=921600,DebugLevel=none -ide-version=10805 -build-path C:\Users\Alberto\AppData\Local\Temp\arduino_build_404925 -warnings=default -build-cache C:\Users\Alberto\AppData\Local\Temp\arduino_cache_722578 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32\examples\ChipID\GetChipID\GetChipID.ino
Using board 'esp32doit-devkit-v1' from platform in folder: C:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32
Using core 'esp32' from platform in folder: C:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32
Detecting libraries used...
"C:\Users\Alberto\AppData\Local\Arduino15\packages\Simba\tools\xtensa-esp32-elf-gcc\1.22.0-59/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_trace" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/heap" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/jsmn" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls_port" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/sdmmc" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spiffs" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wear_levelling" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/console" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/soc" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/coap" "-IC:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wpa_suppli [... cut ...]
             ASSERT_IF_DPORT_REG((_r), REG_SET_BIT);                                                                    \

             ^

C:\Users\Alberto\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/soc/soc/io_mux_reg.h:110:5: note: in expansion of macro 'REG_SET_BIT'

     REG_SET_BIT(PIN_NAME, FUN_PD);

     ^

exit status 1
Errore durante la compilazione per la scheda DOIT ESP32 DEVKIT V1.
Trovata libreria non valida in C:\Users\Alberto\Documents\Arduino\libraries\Releases: C:\Users\Alberto\Documents\Arduino\libraries\Releases`

stickbreaker, 2017-12-03:
@albydnc looking through the error,  that first assert
```
 _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
```
should not be compiled.

It is surrounded by a conditional define that show prevent it from being compiled.
```
 #ifndef __cplusplus
```

I don't have an answer.

Chuck.

albydnc, 2017-12-31:
Still not working

everslick, 2018-04-19:
This issue is closed, because it looks as if it is not a bug or problem with the ESP32 Arduino core or its support libraries. For general API usage questions or help on specific coding challenges, please visit the [arduino-esp32 Gitter channel](https://gitter.im/espressif/arduino-esp32). If you feel this issue was closed in error, reopen it and comment, why you think this is a bug in the Arduino-Core.

closed 2018-04-19 by everslick

albydnc, 2018-04-19:
I have some of my colleagues having the same problem I stated above. Everyone is using a Windows 10 machine with the last Arduino IDE (1.8.3).