espressif/arduino-esp32 #753: Serials don't work in AP

jcv00, 2017-10-21:
### Hardware:
Board:							DOIT ESP32 Dev 1 Module
Core Installation/update date:			11/jul/2017
IDE name:							Arduino IDE 
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
Hi,
I have a problem when run in AP mode, the serial1, serial2 don't receive data, no problem in STA mode.
Any idea?

Thanks

### Sketch:

//Change the code below by your sketch
#include <Arduino.h>

#include <WiFi.h>
#include <HardwareSerial.h>

HardwareSerial Serial2(2);

const char *ssid = "ESP32ap";
const char *password = "12345678";

int c = 0;

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("Configuring access point...");
  /* You can remove the password parameter if you want the AP to be open. */
  WiFi.softAP(ssid, password);

  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
  delay(5000);

  Serial2.begin(9600);
  
}

void loop() {
Serial.print("Serial print: ");
  while (c <= 10 ){
    Serial.print(Serial2.read());
    c++;
  }
}
```

### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:812
load:0x40078000,len:0
load:0x40078000,len:10164
entry 0x400789f8

Configuring access point...AP IP address: 192.168.4.1
[D][WiFiGeneric.cpp:258] _eventCallback(): Event: 13 - AP_STOP

Serial print: -1-1-1-1-1-1-1-1-1-1-1

```

stickbreaker, 2017-10-21:
What is connected to pin 16, 17?
Also you set the Serial2 baud to 9600, one byte would take (1/9600)*10 seconds or (1.04ms).
That tight while loop:
```c++
while (c <= 10 ){
 Serial.print(Serial2.read());
 c++;
 }
```
might complete before a single character could move through the hardware.
with a 240mhz clock rate, the processor could execute 250k instructions before one byte moved.
```c++
uint32_t timeout=millis();
while((millis()-timeout)<1000){ // loop for one second
  if(Serial2.available()){ there is a byte waiting
    int i=Serial2.read();
    Serial.printf("%d %c (0x%02x) Byte available at %ldms\n",i,i,i,millis());
    } 
}

Chuck.

jcv00, 2017-10-21:
is connected a GPS. The reading problem in Serial2, only appears when is in AP mode, not in STA mode.

ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:812
load:0x40078000,len:0
load:0x40078000,len:10164
entry 0x400789f8
ESP32 Chip ID = 1CDA81C40A24
direccion IP AP: 0.0.0.0
[D][WiFiGeneric.cpp:258] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:258] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:258] _eventCallback(): Event: 7 - STA_GOT_IP
IP Address: 192.168.1.103
Wifi Conectado 
UDP iniciado 
$GPGGA,201404.679,4025.2848,N,00339.3400,W,0,0,,628.3,M,51.7,M,,*5D
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,3,1,09,13,71,350,26,20,53,279,,05,52,187,,28,46,072,*7D
$GPGSV,3,2,09,15,45,305,25,30,28,054,,24,18,248,,21,09,301,*74

jcv00, 2017-10-22:
#include <WiFi.h>
#include <HardwareSerial.h>

HardwareSerial Serial2(2);

const char *ssid = "esp32AP";

int c = 0;

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("Configuring access point...");

  WiFi.softAP(ssid);
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);

  Serial2.begin(9600); //GPS connected

}

void loop() {

  if (Serial2.available()) {
    char i = Serial2.read();
    Serial.print(i);
  }
}

I modified the loop() in the sketch with the same result:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:812
load:0x40078000,len:0
load:0x40078000,len:10164
entry 0x400789f8

Configuring access point...AP IP address: 192.168.4.1
[D][WiFiGeneric.cpp:258] _eventCallback(): Event: 13 - AP_STOP

It is strange that when starting the AP it appears as AP_STOP, and the Serial2 port does not recive data. No problem with Wifi in STA mode.
It looks like the serial is deactivated when the AP starts.

jcv00, 2017-10-22:
After testing, I have seen that the problem was in 3.3v power supply from the board, I switched to the vin pins and now it works properly.

Thanks for your help.

closed 2017-10-22 by jcv00

copercini, 2017-10-22:
@jcv00 c'mon man let's update this core, in the new versions it has a brownout (low voltage) detector what help avoid problems like this =)

jcv00, 2017-10-22:
Thanks,
How I can update the new core?

copercini, 2017-10-22:
Here is a tutorial: https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md#how-to-update-to-the-latest-code

jcv00, 2017-10-22:
Thanks, I have the last code.