espressif/arduino-esp32 #755: Onewire & Wifi Problems

Northernboy, 2017-10-22:
Whenever I use Onewire & Wifi together on the ESP32, the readings from my DS18B20 temperature sensor are erratic. As soon I disable the Wifi the temperature sensor reads normal.
Just wonder if anyone else has seen this?

stickbreaker, 2017-10-22:
I would suspect that the bitbanging time sensitive protocol of OneWire is being disrupted by the background WiFi process,  It is being preempted during it's critical sections. 
I don't know if it is possible to disable the interrupts long enough to communicate with the DS18B20.

Chuck.

pieman64, 2017-10-22:
@Northernboy can you provide precise data for "erratic"?

Northernboy, 2017-10-23:
With the Wifi disabled reading I get are proper, steady & CRC checks ok. With the Wifi enabled reading are out of range eg.-127, I get one good reading in 50, the CRC does not agree. 
Hopefully this explains it.

me-no-dev, 2017-10-23:
I imagine the OneWire lib needs to be rewritten to support ESP32. Since ESP32 runs on OS, the task reading the 1W interface will be interrupted and timing will be wrong. There are some peripherals that can be used ;) like RMT

CelliesProjects, 2017-10-24:
@Northernboy I am using the OneWire lib 2.3.3 together with WiFi in one of my projects. Seems to be working fine. A few CRC errors, I would say about 1 every 10-20 secs.
Could you show some code?

Northernboy, 2017-10-25:
Here is a simplified version of the code I am using to read the DS18B20s.
Running the code this way will give CRC fails and wrong temperature readings.
If comment out the Wifi commands, I don't get any CRC fails and temperature reads correct.

@me-no-dev I started looking into the RMT but could not find much documentation on it for Arduino or an example but not giving up.

Thank you for all your inputs and ideas.

```ruby
#include <OneWire.h>
#include <WiFi.h>
#include <SimpleTimer.h>

SimpleTimer timer;
OneWire ds(15);

const char* ssid = "###";
const char* password = "#############";

int ledState = LOW;             // ledState used to set the LED

byte Sensor[8] = {0x28,0xFF,0x99,0x39,0x68,0x14,0x03,0x62};
byte Outside[8] ={0x28,0xFF,0x99,0x56,0x68,0x14,0x03,0x68};

float temp = 18;
float out = 0;

int temp_count[4]={0,0,0,0};

void setup() {
  Serial.begin(115200);
  delay(5000);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  
  timer.setInterval(5000, read_in);
  timer.setInterval(10222, read_out);
  timer.setInterval(500, LED);
  
  init_temp(Sensor);
  init_temp(Outside);

  pinMode(13, OUTPUT);
  digitalWrite(13, ledState);
}

void loop() {
  timer.run();  
}

void LED(void){
  ledState = !ledState;
  digitalWrite(13, ledState);
}
    
float getTemp(byte add[8], byte index){
  byte i, dat;
  byte data[12];
  temp_count[index]++;
  long start=millis();
  ds.reset();
  ds.select(add);
  ds.write(0x44,1);
  dat = ds.read();  
  while (!dat){
  dat = ds.read();}  
  dat = ds.reset();  
  ds.select(add);  
  ds.write(0xBE);
  
  for ( i = 0; i < 9; i++) {
  data[i] = ds.read();}
  
  if(OneWire::crc8(data, 8)!=data[8]){temp_count[index+1]++;Serial.println("CRC Fail");}
  
  int16_t raw = (data[1] << 8) | data[0];
  byte cfg = (data[4] & 0x60);
  
  // at lower res, the low bits are undefined, so let's zero them
  if (cfg == 0x00) raw = raw & ~7;  // 9 bit resolution, 93.75 ms
  else if (cfg == 0x20) raw = raw & ~3; // 10 bit res, 187.5 ms
  else if (cfg == 0x40) raw = raw & ~1; // 11 bit res, 375 ms
  //// default is 12 bit resolution, 750 ms conversion time
  
  start=millis()-start;
  Serial.println(start);
  //Serial.print("Inside ");Serial.print(temp_count[0]);Serial.print(":");Serial.println(temp_count[1]);
  //Serial.print("Out    ");Serial.print(temp_count[2]);Serial.print(":");Serial.println(temp_count[3]);
  Serial.println((float)raw / 16.0);
  Serial.println(" ");
  return (float)raw / 16.0;
    
}

void init_temp(byte add[8]){
  ds.reset();
  ds.select(add);
  ds.write(0x4E);         // Write scratchpad
  ds.write(0);            // TL
  ds.write(0);            // TH
  ds.write(0x3F);         // Configuration Register 12bit=0X7F 11bit=0X5F 10bit= 0X3F
  ds.write(0x48); 
}

void read_in(void){
  Serial.println("Inside");
  temp=getTemp(Sensor,0);
}

void read_out(void){
  Serial.println("Outside");
  out=getTemp(Outside,2);
}
```

CelliesProjects, 2017-10-27:
I never included 'SimpleTimer.h' but use Dallas sensors and Wifi together without problem.
 
This works for me:

````
/*************************************************************************************************
 * 
 *     Read multiple DS18B20 Dallas sensors in a task.
 *     
 *     For ESP32 by CelliesProjects 2017.
 *     
 *     Code adapted from https://www.pjrc.com/teensy/td_libs_OneWire.html
 * 
 ************************************************************************************************/
//#define SHOW_DALLAS_ERROR        // uncomment to show Dallas ( CRC ) errors on Serial.
#define ONEWIRE_PIN           5    // OneWire Dallas sensors are connected to this pin
#define MAX_NUMBER_OF_SENSORS 3    // maximum number of Dallas sensors

#include "OneWire.h" 
#include <WiFi.h>

const char* ssid = "----------";
const char* password = "----------";

OneWire  ds( ONEWIRE_PIN );        // (a 4.7K pull-up resistor is necessary)

struct sensorStruct
{
  byte addr[8];
  float temp;
  String name;
} sensor[MAX_NUMBER_OF_SENSORS];

byte numberOfFoundSensors;

void setup()
{
  Serial.begin( 115200 );
  Serial.println( "\n\nMultiple DS18B20 sensors as task ESP32 example." );
  
  Serial.printf( "Connecting to %s with password %s\n", ssid,  password );

  xTaskCreatePinnedToCore(
    tempTask,                       /* Function to implement the task */
    "tempTask ",                    /* Name of the task */
    4000,                           /* Stack size in words */
    NULL,                           /* Task input parameter */
    5,                              /* Priority of the task */
    NULL,                           /* Task handle. */
    1);                             /* Core where the task should run */
    
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while ( WiFi.status() != WL_CONNECTED )
  {
    vTaskDelay( 250 /portTICK_PERIOD_MS );
    Serial.print( "." );
  } 
  Serial.println();
}

void loop()
{
  if ( numberOfFoundSensors )
  {
    Serial.println( String( millis() / 1000.0 ) + " sec" );
    for ( byte thisSensor = 0; thisSensor < numberOfFoundSensors; thisSensor++ )
    {
      Serial.println( sensor[thisSensor].name + ": " + String( sensor[thisSensor].temp / 16.0 ) + "°C" );
    }
  }
  else
  {
    Serial.println( "No Dallas sensors." );
  }
  vTaskDelay( 500 / portTICK_PERIOD_MS );
}

void tempTask( void * pvParameters )
{
  numberOfFoundSensors = 0;
  byte currentAddr[8];
  while ( ds.search( currentAddr ) && numberOfFoundSensors < MAX_NUMBER_OF_SENSORS )
  {
    //Serial.write( "Sensor "); Serial.print( counter ); Serial.print( ":" );
    for ( byte i = 0; i < 8; i++)
    {
      //Serial.write(' ');
      //Serial.print( currentAddr[i], HEX );
      sensor[numberOfFoundSensors].addr[i] = currentAddr[i];
    }
    //sensor[numberOfFoundSensors].name = 'T ' + char( numberOfFoundSensors );
    numberOfFoundSensors++;
  }
  Serial.printf( "%i Dallas sensors found.\n", numberOfFoundSensors );

  if ( !numberOfFoundSensors )
  {
    vTaskDelete( NULL );
  }

  /* main temptask loop */

  while (1)
  {
    for ( byte thisSensor = 0; thisSensor < numberOfFoundSensors; thisSensor++)
    {
      ds.reset();
      ds.select( sensor[thisSensor].addr );
      ds.write( 0x44, 0);        // start conversion, with parasite power off at the end
    }

    vTaskDelay( 750 / portTICK_PERIOD_MS); //wait for conversion ready

    for ( byte thisSensor = 0; thisSensor < numberOfFoundSensors; thisSensor++)
    {
      byte data[12];
      ds.reset();
      ds.select( sensor[thisSensor].addr );
      ds.write( 0xBE );         // Read Scratchpad

      //Serial.print( "Sensor " );Serial.print( thisSensor ); Serial.print("  Data = ");
      //Serial.println( present, HEX );
      //Serial.print(" ");
      for ( byte i = 0; i < 9; i++)
      { // we need 9 bytes
        data[i] = ds.read(  );
        //Serial.print(data[i], HEX);
        //Serial.print(" ");
      }
      //Serial.println();

      byte type_s;
      // the first ROM byte indicates which chip
      switch ( sensor[thisSensor].addr[0] )
      {
        case 0x10:
          //Serial.println("  Chip = DS18S20");  // or old DS1820
          type_s = 1;
          break;
        case 0x28:
          //Serial.println("  Chip = DS18B20");
          type_s = 0;
          break;
        case 0x22:
          //Serial.println("  Chip = DS1822");
          type_s = 0;
          break;
        default:
#ifdef SHOW_DALLAS_ERROR
          Serial.println("Device is not a DS18x20 family device.");
#endif
          return;
      }

      int16_t raw;
      if ( OneWire::crc8(data, 8) != data[8])
      {
#ifdef SHOW_DALLAS_ERROR
        // CRC of temperature reading indicates an error, so we print a error message and discard this reading
        Serial.print( millis() / 1000.0 ); Serial.print( " - CRC error from device " ); Serial.println( thisSensor );
#endif
      }
      else
      {
        raw = (data[1] << 8) | data[0];
        if (type_s)
        {
          raw = raw << 3; // 9 bit resolution default
          if (data[7] == 0x10)
          {
            // "count remain" gives full 12 bit resolution
            raw = (raw & 0xFFF0) + 12 - data[6];
          }
        }
        else
        {
          byte cfg = (data[4] & 0x60);
          // at lower res, the low bits are undefined, so let's zero them
          if (cfg == 0x00) raw = raw & ~7;  // 9 bit resolution, 93.75 ms
          else if (cfg == 0x20) raw = raw & ~3; // 10 bit res, 187.5 ms
          else if (cfg == 0x40) raw = raw & ~1; // 11 bit res, 375 ms
          //// default is 12 bit resolution, 750 ms conversion time
        }
        sensor[thisSensor].temp = raw;
      }
    }
  }
}
````

If it works for you too, I would see if dropping the 'SimpleTimer.h' maybe solves the problem.

CelliesProjects, 2017-10-27:
A little expanded example with a simple webserver included:

```c
/*************************************************************************************************
 * 
 *     Simple DS18B20 Dallas sensor webserver.
 *     
 *     For ESP32 by CelliesProjects 2017.
 *     
 *     Code adapted from https://www.pjrc.com/teensy/td_libs_OneWire.html
 * 
 ************************************************************************************************/
//#define SHOW_DALLAS_ERROR        // uncomment to show Dallas ( CRC ) errors on Serial.
#define ONEWIRE_PIN           5    // OneWire Dallas sensors are connected to this pin
#define MAX_NUMBER_OF_SENSORS 3    // maximum number of Dallas sensors

#include "OneWire.h"
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

const char* ssid = "----------";
const char* password = "-----------";

AsyncWebServer server(80);

OneWire  ds( ONEWIRE_PIN );        // (a 4.7K pull-up resistor is necessary)

struct sensorStruct
{
  byte    addr[8];
  float   tempCelcius;
  char    name[8];
} sensor[MAX_NUMBER_OF_SENSORS];

byte numberOfFoundSensors;

void setup()
{
  Serial.begin( 115200 );
  Serial.println( "\n\nSimple DS18B20 Dallas sensor webserver example.\n" );
  
  xTaskCreatePinnedToCore(
    tempTask,                       /* Function to implement the task */
    "tempTask ",                    /* Name of the task */
    4000,                           /* Stack size in words */
    NULL,                           /* Task input parameter */
    5,                              /* Priority of the task */
    NULL,                           /* Task handle. */
    1);                             /* Core where the task should run */
    
  Serial.printf( "Connecting to '%s' with password '%s'\n", ssid,  password );
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while ( WiFi.status() != WL_CONNECTED )
  {
    vTaskDelay( 250 /portTICK_PERIOD_MS );
    Serial.print( "." );
  } 
  Serial.println();

  server.on( "/", HTTP_GET, []( AsyncWebServerRequest * request )
  {
    if ( !numberOfFoundSensors )
    {
      return request->send( 501, "text/plain", "No sensors." );
    }

    char content[200];
    uint8_t usedChars = 0;
[... 127 lines cut ...]
        raw = (data[1] << 8) | data[0];
        if (type_s)
        {
          raw = raw << 3; // 9 bit resolution default
          if (data[7] == 0x10)
          {
            // "count remain" gives full 12 bit resolution
            raw = (raw & 0xFFF0) + 12 - data[6];
          }
        }
        else
        {
          byte cfg = (data[4] & 0x60);
          // at lower res, the low bits are undefined, so let's zero them
          if (cfg == 0x00) raw = raw & ~7;  // 9 bit resolution, 93.75 ms
          else if (cfg == 0x20) raw = raw & ~3; // 10 bit res, 187.5 ms
          else if (cfg == 0x40) raw = raw & ~1; // 11 bit res, 375 ms
          //// default is 12 bit resolution, 750 ms conversion time
        }
        sensor[thisSensor].tempCelcius = raw / 16.0;
      }
    }
  }
}
```

Northernboy, 2017-10-27:
Thank you I'll test it next week. I noticed that you are using code to separate the tasks for the processors, I was looking into this to see how it was done but you gave me a good example. I am confident it will work better.
Thanks again and I'll report back after my testing.

theuselessone, 2017-10-30:
I have exactly the same problem. If i comment out ConnectWifi(); then the readings from the sensor is OK otherwise i get -127. I will also try your example.

`/********************************************************************/
// First we include the libraries
#include <OneWire.h>
#include <DallasTemperature.h>
#include <WiFi.h>

/********************************************************************/
// Data wire is plugged into pin 14 on the ESP32

#define ONE_WIRE_BUS 14
/********************************************************************/
const char* ssid     = "ASUS";
const char* password = "";

WiFiServer server(80);
/********************************************************************/
/********************************************************************/
// Setup a oneWire instance to communicate with any OneWire devices
// (not just Maxim/Dallas temperature ICs)
OneWire oneWire(ONE_WIRE_BUS);
/********************************************************************/
// Pass our oneWire reference to Dallas Temperature.
DallasTemperature sensors(&oneWire);
/********************************************************************/
void setup(void)
{
  // start serial port
  Serial.begin(115200);
  delay(3000); 

  Serial.println("Dallas Temperature IC Control Library Demo");
  // Start up the library
  sensors.begin();

  // ******************************
  // We start by connecting to a WiFi network
  // ******************************
  //
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  // ******************************
  // If i comment out this the sensor work otherwise i just get -127.0
  // ******************************
  ConnectWifi(); // If i comment out this the sensor work otherwise i just get -127.0

}
void loop(void)
{

  // request to all devices on the bus
  /********************************************************************/
  Serial.print(" Requesting temperatures...");
  sensors.requestTemperatures(); // Send the command to get temperature readings
  Serial.println("DONE");
  /********************************************************************/
  Serial.print("Temperature is: ");
  Serial.print(sensors.getTempCByIndex(0)); // Why "byIndex"?
  // You can have more than one DS18B20 on the same bus.
  // 0 refers to the first IC on the wire
  delay(3000);
  Serial.println("In loop ");
}

void ConnectWifi()
{

  // attempt to connect to Wifi network:
  while ( WiFi.status() != WL_CONNECTED) {
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);
    // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
    WiFi.begin(ssid, password);

    // wait 10 seconds for connection:
    delay(10000);
  }

}
`

theuselessone, 2017-10-31:
I changed my code to implement CelliesProjects example and that works for me. Thank you for sharing! 

A stupid question perhaps you will have to excuse a novice but sensor[XX].name is something that i define or?

Northernboy, 2017-11-03:
I tested the above projects and they worked, then I researched programming the Cores and tasks in FreeRTOS. Everything worked however I still got the odd failure reading a temperature, like 1 in 50 which isn't bad. Then I tested stopping & starting interrupts when I was reading the temperature sensors and that worked 100%. Went back to my original code without tasks & core programming and added the same interrupt stop & start and it worked perfect.

Here is an example of my code:

```
portDISABLE_INTERRUPTS();
temp=read_temperature();
portENABLE_INTERRUPTS();
```

Thanks again for all your help and hopefully it will help someone else.

cristianhumelnicu, 2017-11-05:
theuselessone , Northernboy

Try to move sensor reading in  "void setup" from void loop.... and all will be ok

stickbreaker, 2017-11-05:
Has anyone looked at writing code for the RTC_CPU to handle this sensor directly?  I would think It would be a perfect solution. The RTC_CPU currently is idle.  If someone wrote a routine to handle OneWire with a dedicated pin, you could send a string of OneWire commands which the RTC_CPU could execute in real time without having to worry about background RTOS processes interrupting the critical timing sequences.

Chuck.

cristianhumelnicu, 2017-11-27:
@Northernboy Hi , could you share your  code with start/stop intrerupts ? Thx!

Northernboy, 2017-11-28:
@cristianhumelnicu , as I posted up above
```
portDISABLE_INTERRUPTS();
temp=read_temperature();
portENABLE_INTERRUPTS();
```
portDISABLE_INTERRUPTS() is called before you do the temperature sensor read however its done in your code.

portENABLE_INTERRUPTS() is called after the temperature sensor read.

Hope this helps, as it has helped me.

stickbreaker, 2017-12-29:
@Northernboy  @cristianhumelnicu I think you should use `portENTER_CRITICAL(&mux)`.  Disabling interrupts doesn't stop the task scheduler from switching active tasks.  I modified the standard Arduino OneWire library to use CRITICAL sections.  It has been working successfully with DS18b20's 

[stickbreaker/OneWire](https://github.com/stickbreaker/OneWire)

Chuck.

lchamps, 2017-12-30:
Hello stickbreaker
Thanks a lot and happy new year!
a little bit when I use DallasTemperature.h :

float getTemp()
{
  sensors.requestTemperatures();
  delay(400); // or other value - without this I get 85 in parasitic mode
  return sensors.getTempCByIndex(0);
}

Luiz

stickbreaker, 2017-12-30:
@lchamps The Dallas DS18b20 can take up to 750ms to do a conversion cycle. The conversion time is dependent on resolution 9 .. 12 bits. When the device completes the sample it stores the reading in the 'scratch pad' memory.  I suspect your `sensors.getTempCByIndex()` just reads the scratchpad. 

I wrote a function to encapsulate this delay:

```c++
typedef struct {
  uint8_t id[8];
  uint8_t pin;
  uint32_t timeout;
} ONEWIRE_t;

ONEWIRE_t sensor;

void initSensor(ONEWIRE_t &sensor){
sensor.id = {0x28, ... };
sensor.pin =17;
sensor.timeout=0;
}

int16_t getTemp(ONWIRE_t sensor){
if(sensor.timeout==0){
// send temp conversion command
  sensor.timeout = millis(); // start timeout
  return -32768; // conversion in progress
  }
else if(millis()-sensor.timeout>1000){// at least one second since conversion command
  int16_t temp=0; // temp will be formatted as an implied 2 decimal value -50.00 .. +125.00 degree C  
// read scratchpad
 sensor.timeout=0; // ready for next conversion request
 return temp;
  }
else return -32768; // timeout has not expired
}
```
when ever I want the temperature I use this:

```c++
int16_t temp = getTemp(sensor);
if(temp!=-32768){// got good temp
  }
else {// still in conversion, try again later
}
```

These coding examples are offhand, so they probably have syntax issues.

Chuck.

JacoFourie, 2018-02-13:
OK I have the same issues. If I use the Library from Chuck I get values but every 2 to 3 will be an error. If I disable the interrupts then I get no errors and I get valid values all the time and wifi works 100 % but then I loose SPIFFS. I save the html in SPIFFS. So how do we get al 3 parts to work together 100% ?

JacoFourie, 2018-02-13:
Here is the code I am using.

````

#include <WiFi.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <WebSocketsServer.h>
#include <DallasTemperature.h>
#include <OneWire.h>
#include <RunningMedian.h>
#include <PubSubClient.h>
#include <WiFiClientSecure.h>
#include <UniversalTelegramBot.h>
#include <WebServer.h>
#include <SPIFFS.h>
#include "FS.h"

#define COMMAND_START_CHAR  'U'
#define COMMAND_END_STRING  "\r\n"
int cmdEndStrLen = strlen(COMMAND_END_STRING);
String cmdBuf = "";

TaskHandle_t wifi_tasks;
SemaphoreHandle_t baton;

int feeder_relay_pin = 32;
int stoker_relay_pin = 33;

OneWire oneWire_1(18);
DallasTemperature sensor_t1(&oneWire_1);

RunningMedian s_temp = RunningMedian(11); //Smooth Stoker Temp probe 1
float temp_c = 0, old_temp = 0, temp_value = 0;
int sensor_loop = 0;
int wificounter = 0;
char desp_temp[6];
long rssi;
long wifi_quality;

const char* ssid     = "Pandora-studio";
const char* password = "xxx";
const char* mqtt_server = "10.64.9.34";

//String WebPage = "<!DOCTYPE html><html><style>input[type=\"text\"]{width: 90%; height: 3vh;}input[type=\"button\"]{width: 9%; height: 3.6vh;}.rxd{height: 90vh;} textarea{width: 99%; height: 100%; resize: none;}</style><script>var Socket;function start(){Socket=new WebSocket('ws://' + window.location.hostname + ':81/'); Socket.onmessage=function(evt){ document.getElementById(\"rxConsole\").value +=evt.data; }}function enterpressed(){Socket.send(document.getElementById(\"txbuff\").value); document.getElementById(\"txbuff\").value=\"\";}</script><body onload=\"javascript:start();\"> <div><input class=\"txd\" type=\"text\" id=\"txbuff\" onkeydown=\"if(event.keyCode==13) enterpressed();\"><input class=\"txd\" type=\"button\" onclick=\"enterpressed();\" value=\"Send\" > </div><br><div class=\"rxd\"> <textarea id=\"rxConsole\" readonly></textarea> </div></body></html>";
//String WebPage = "<!DOCTYPE html><html><style>input[type=\"text\"]{width: 90%; height: 3vh;}input[type=\"button\"]{width: 9%; height: 3.6vh;}.rxd{height: 90vh;}textarea{width: 99%; height: 100%; resize: none;}</style><script>var Socket;function start(){Socket=new WebSocket('ws://' + window.location.hostname + ':1886/'); Socket.onmessage=function(evt){ document.getElementById(\"rxConsole\").value +=evt.data; document.getElementById(\"temp_val\").value=evt.data; }}function enterpressed(){Socket.send(\"Hallo Jaco. You pressed the button !!\"); document.getElementById(\"txbuff\").value=\"\";} function relayon(){Socket.send(\"U1\");} function relayoff(){Socket.send(\"U2\");}   </script><body onload=\"javascript:start();\"> <div><input class=\"txd\" type=\"text\" id=\"txbuff\" onkeydown=\"if(event.keyCode==13) enterpressed();\"><input class=\"txd\" type=\"button\" onclick=\"enterpressed();\" value=\"Send\" ></div> <div><input type = \"text\" id = \"temp_val\" value = \"0.0\" /></div><br><div class=\"rxd\"> <input class=\"txd\" type=\"button\" onclick=\"relayon();\" value=\"Relay On\" > <input class=\"txd\" type=\"button\" onclick=\"relayoff();\" value=\"Relay Off\" >   <textarea id=\"rxConsole\" readonly></textarea> </div></body></html>";
//String WebPage = "<!DOCTYPE html><html><style>input[type=\"text\"]{width: 90%; height: 3vh;}input[type=\"button\"]{width: 9%; height: 3.6vh;}.rxd{height: 90vh;}textarea{width: 99%; height: 100%; resize: none;}</style><script>const WebSocket = require('ws'); var wsc = new WebSocketClient();function start(){wsc.open('ws://' + window.location.hostname + ':1886/');wsc.onmessage = function(data,flags,number){console.log(`WebSocketClient message #${number}: `,data); document.getElementById(\"rxConsole\").value += data; document.getElementById(\"temp_val\").value = data; } function enterpressed(){ wsc.send(\"Hallo Jaco. You pressed the button !!\"); document.getElementById(\"txbuff\").value=""; } function relayon(){ wsc.send(\"U1\");} function relayoff(){ wsc.send(\"U2\");} function WebSocketClient(){ this.number = 0; this.autoReconnectInterval = 5*1000;  } WebSocketClient.prototype.open = function(url){ this.url = url; this.instance = new WebSocket(this.url); this.instance.on('open',()=>{ this.onopen(); }); this.instance.on('message',(data,flags)=>{this.number ++; this.onmessage(data,flags,this.number);});this.instance.on('close',(e)=>{switch (e){case 1000: console.log(\"WebSocket: closed\");break;default: this.reconnect(e);break;}this.onclose(e);});this.instance.on('error',(e)=>{switch (e.code){case 'ECONNREFUSED':this.reconnect(e);break;default:this.onerror(e);break;}});}WebSocketClient.prototype.send = function(data,option){try{this.instance.send(data,option);}catch (e){this.instance.emit('error',e);}}WebSocketClient.prototype.reconnect = function(e){console.log(`WebSocketClient: retry in ${this.autoReconnectInterval}ms`,e);this.instance.removeAllListeners();var that = this;setTimeout(function(){console.log(\"WebSocketClient: reconnecting...\");that.open(that.url);},this.autoReconnectInterval);}WebSocketClient.prototype.onopen = function(e){ console.log(\"WebSocketClient: open\",arguments); }WebSocketClient.prototype.onmessage = function(data,flags,number){  console.log(\"WebSocketClient: message\",arguments);}WebSocketClient.prototype.onerror = function(e){  console.log(\"WebSocketClient: error\",arguments);}WebSocketClient.prototype.onclose = function(e){  console.log(\"WebSocketClient: closed\",arguments);}</script><body onload=\"javascript:start();\"> <div><input class=\"txd\" type=\"text\" id=\"txbuff\" onkeydown= [... cut ...]
void processCommand(String cmd) {
  if (cmd.length() > 1) {
    byte cmdId = cmd.charAt(1);
    switch (cmdId) {
      //U1 Switch Relays On
      case 0x31:
        digitalWrite(feeder_relay_pin, HIGH);
        digitalWrite(stoker_relay_pin, HIGH);
        Serial.println("Relay on");
        break;
      //U2  Switch Relays Off
      case 0x32:
        digitalWrite(feeder_relay_pin, LOW);
        digitalWrite(stoker_relay_pin, LOW);
        Serial.println("Relay off");
        break;
      
    }
  }
}
````

bill-orange, 2018-11-12:
Well, It thought this issue was dead.  I have been using the code posted by @CelliesProjects for well over a year.  Dallas sensors worked flawlessly until I recompiled yesterday.  Apparently some change in a library over the past year has affected the code.  Dallas sensors are no longer found.  Can someone confirm?

stickbreaker, 2018-11-12:
@bill-orange I rewrote a one-wire library for esp32 [onewire](https://github.com/stickbreaker/OneWire)

Try it. It works for me,

Chuck

bill-orange, 2018-11-12:
Yes, but is it compatible with Wifi?  That seems to be the issue.  WiFi used with Dallas gives a bad reading about one-in-fifty times unless special precautions are taken.  That problem was licked but the code that I mentioned in my post (with the special precautions) no longer seems to work on a fresh compile.   I suspect a change in OneWire.  Trouble is I have not recompiled for over a year so I don’t know when the change happened. Plus I could be screwed up.  Some should confirm what I am finding.

From: chuck todd 
Sent: Monday, November 12, 2018 9:05 AM
To: espressif/arduino-esp32 
Cc: bill-orange ; Mention 
Subject: Re: [espressif/arduino-esp32] Onewire & Wifi Problems (#755)

@bill-orange I rewrote a one-wire library for esp32 onewire

Try it. It works for me,

Chuck

bill-orange, 2018-11-12:
Okay, mostly a false alarm.  I had a low battery.  So perfectly low that the Dallas chips would read if WiFi was turned off.  

With that problem solved, it still seems unfortunate that here is no Dallas_32.h library function that would avoid all this.  @CelliesProjects code works but it is awfully awkward just to get the temperature.

@stickbreaker code is a lot more compact but without  the  xTaskCreatePinnedToCore stuff I am pretty sure that it would suffer the occasional error.

In my humble opinion should task should be added to the development road map if it is not here already.  I am no volunteering!  Beyond my skill.

dforsyth1, 2019-03-08:
Had same issue (OneWire vs Wifi).  I know OneWire is completely time sensitive but lacked the FreeRTOS skills..  Tried stickbreakers ESP32 OneWIre library and it works -no issues at all now!!  

Many thanks for sharing.  :)

bill-orange, 2019-03-16:
I suppose that it might have something to do with switching to the 1.0.1 Core but I still find that to avoid bad data I have to pin the OneWire function to a single core using Vtask.  This is even using @stickbreaker fork.  Rock solid with his fork and vTask.  How much this has to do with WiFi I don't know.

closed 2019-08-15 by stale[bot]