espressif/arduino-esp32 #758: WiFi.scanNetworks() does not work with last version of repository

mikey60, 2017-10-22:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		Clean install on windows October 21 2017
IDE name:						Arduino IDE
Flash Frequency:					40 and  80Mhz tried
Upload Speed:					921600?

### Description:
After doing a clean install of the the latest arduino-esp32 the WiFi.scanNetworks() function does not work any more in my program so I tried the ESP32 WiFIScan.ino example and the serial monitor reports:

scan start
scan done
no networks found

It worked correctly in an earlier version of arduino-esp32 (version of approximately 2 weeks ago)
### Sketch:
WiFIScan.ino
```
/*
 *  This sketch demonstrates how to scan WiFi networks.
 *  The API is almost the same as with the WiFi Shield library,
 *  the most obvious difference being the different file you need to include:
 */
#include "WiFi.h"

void setup()
{
    Serial.begin(115200);

    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);

    Serial.println("Setup done");
}

void loop()
{
    Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found
    int n = WiFi.scanNetworks();
    Serial.println("scan done");
    if (n == 0) {
        Serial.println("no networks found");
    } else {
        Serial.print(n);
        Serial.println(" networks found");
        for (int i = 0; i < n; ++i) {
            // Print SSID and RSSI for each network found
            Serial.print(i + 1);
            Serial.print(": ");
            Serial.print(WiFi.SSID(i));
            Serial.print(" (");
            Serial.print(WiFi.RSSI(i));
            Serial.print(")");
            Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
            delay(10);
        }
    }
    Serial.println("");

    // Wait a bit before scanning again
    delay(5000);
}
```

### Debug Messages:
```
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 1 - SCAN_DONE

```

derpicknicker1, 2017-10-22:
This is a known bug. See #242

mikey60, 2017-10-22:
Thank you for the tip.  I added the two following lines of code before the WiFi.scanNetworks() function as suggested at the end of #242 and it worked. 
```
    WiFi.disconnect();
    WiFi.begin("lolllool","loooooooool"); //Can be a nonexistent network
```

closed 2017-10-22 by mikey60