espressif/arduino-esp32 #759: PietteTech DHT on ESP32

SayedBaqer, 2017-10-22:
Hi,
i am using PietteTech DHT library [on](https://github.com/chaeplin/PietteTech_DHT-8266) ESP8266 without any problem, but when i try to use it on ESP32 it show error (PietteTech_DHT-8266-master\PietteTech_DHT.cpp:49:28: fatal error: user_interface.h: No such file or directory)
i try to disable this line from library, then i got again error  (PietteTech_DHT-8266-master\PietteTech_DHT.cpp:84:21: error: expected initializer before 'PietteTech_DHT'

 int ICACHE_RAM_ATTR PietteTech_DHT::acquire() {)
any body can help to get the library to work?

copercini, 2017-10-22:
Try remove the lines that include `user_interface.h` in PietteTech_DHT.cpp
and replace `ICACHE_RAM_ATTR` by `IRAM_ATTR`

SayedBaqer, 2017-10-22:
thanks very much work fine

everslick, 2018-04-19:
This issue is closed, because it looks as if it is not a bug or problem with the ESP32 Arduino core or its support libraries. For general API usage questions or help on specific coding challenges, please visit the [arduino-esp32 Gitter channel](https://gitter.im/espressif/arduino-esp32). If you feel this issue was closed in error, reopen it and comment, why you think this is a bug in the Arduino-Core.

closed 2018-04-19 by everslick

ferlanjo, 2021-04-07:
This occurrence this error in the compilation, has anyone had this problem?

C:\Program Files (x86)\Arduino\libraries\PietteTech_DHT-master/PietteTech_DHT.h:31:25: fatal error: application.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board ESP32 Dev Module.