espressif/arduino-esp32 #761: nvs_open fail error when using WiFi

nkolban, 2017-10-22:
I am using the latest code as of this post.  Both the latest ESP-IDF from Git master and the latest Arduino-ESP32 download.  I have a sketch which enables WiFi ... see:

https://pastebin.com/7KvKX27J

All compiles and flashes cleanly, but at runtime, I get the error:
```
Connecting to XXX
W (84) wifi: wifi nvs_open fail ret=4353

E (84) wifi: wifi_init 1410 ret=4353
[E][WiFiGeneric.cpp:120] wifiLowLevelInit(): esp_wifi_init 4353
W (87) wifi: wifi nvs_open fail ret=4353

E (90) wifi: wifi_init 1417 ret=4353
[E][WiFiGeneric.cpp:120] wifiLowLevelInit(): esp_wifi_init 4353
.......................
```

I am building my solution using Arduino as a component in ESP-IDF as opposed to using the Arduino IDE.  I can recreate at will and am happy to provide any additional information or screen share as needed.

me-no-dev, 2017-10-22:
Your BLE device causing this? I saw in your BLE code that you nvs_open

nkolban, 2017-10-22:
It shouldn't ... if we look at the pastebin of the code I am using here:

https://pastebin.com/7KvKX27J

I had hoped I had commented out all the BLE code calls... so it was nothing but WiFi.  I'm for sure not saying the error isn't in my code ... I'm just not seeing it yet and have tried to back out to just WiFi.  I noticed that the WiFi library changed in Arduino-ESP32 the other day.

igrr, 2017-10-30:
Don't forget to initialize NVS using `nvs_flash_init()` before enabling WiFi.
Previously that was not strictly required, as WiFi driver would also call nvs_flash_init if the application didn't. It doesn't do that anymore, so application should take care of NVS initialization before WiFi or Bluetooth drivers are started.

me-no-dev, 2017-10-30:
WiFi Does start the NVS :) @nkolban is this still present with the latest build? I had no issues running BLE and WiFi at the same time

coolbreeze413, 2017-10-31:
I am also trying to use my custom code with the arduino-esp32 core as an esp-idf component, and ran into the same problem. I found that nvs_flash_init() called from app_main() was failing. The error code was ESP_ERR_NVS_NO_FREE_PAGES indicating a problem with the nvs partition.
After erasing the chip flash with "make erase_flash" the nvs init succeeded, and the WiFi connection also worked (and nvs_open failures were gone).
I am not sure why the problem appeared suddenly, probably some change in the esp-idf/arduino-esp32 triggered this and we need to erase the flash fully to make it work correctly.

me-no-dev, 2017-11-06:
this can happen if you switch partition schemes and they have different nvs size.

kenjican, 2017-11-21:
I encoutered the same problem, the solution is to unselect  Wifi nvs enabled in menuconfig ,component config -> Wifi -> Wifi NVS Flash.  And make flash.

tablatronix, 2018-02-06:
I just had a similar issue out of the blue, no ble.

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11404
entry 0x40078a28
W (17) wifi: wifi nvs_open fail ret=4353

E (17) wifi: wifi_init 1408 ret=4353
[E][WiFiGeneric.cpp:118] wifiLowLevelInit(): esp_wifi_init 4353

 Starting
E (23) wifi: esp_wifi_get_mode 789 wifi is not init
E (23) wifi: esp_wifi_get_channel 1168 wifi is not init
E (23) wifi: esp_wifi_get_auto_connect 1345 wifi is not init
Mode: 
Channel: 252
Auto connect: 63
E (31) wifi: esp_wifi_get_config 1011 wifi is not init
SSID (12): �E�?��?��?
Passphrase (0): 
BSSID set: 1
```

tablatronix, 2018-02-06:
printdiag was cause,  #1088

tablatronix, 2018-03-21:
`W (20) wifi: wifi nvs_open fail ret=4353`

I start getting this again..

Gonna nvs erase

closed 2018-04-19 by everslick