espressif/arduino-esp32 #763: gen_esp32part.exe - thorough a risk analysis engine

ioanvapi, 2017-10-23:
### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
I have installed arduino-esp32 library as described in documentation. The Symantec Antivirus alerts that gen_esp32part.exe is suspect. Some guy at systems ran it through a risk analysis engine and the result is:
Risk Assesment:
Remote Access: Reads terminal service related keys (RDP related)
Persistence: Writes data to a remote process

I had to delete that file from system. Can somebody suggest an workarround or else?

me-no-dev, 2017-10-23:
there s nothing wrong with the file. It;s a compiled version of the python file with same name in the same folder. We have run this through Microsoft as well. I advise you to get the file and add is as an exception in your Symantec, else you will not be able to compile sketches.

closed 2017-10-23 by me-no-dev

ioanvapi, 2017-10-23:
Thanks, I've already done that. It works in a home environment but not in an office/working environment. In an working environment the security trusts their tools first :). Maybe the exe has bundled some libraries code that looks suspect.

me-no-dev, 2017-10-23:
It's nothing special :) it uses the most popular py2bin python packager. Without it every user will need to install python and some packages in order to compile a sketch. That has proven to be an issue ;) Maybe you can use Linux VM and compile in that?