espressif/arduino-esp32 #764: Compile error "cc1plus: internal compiler error: Illegal instruction: 4"

gerardwr, 2017-10-23:
### Hardware:
Board:						ESP32 Dev Module?
Core Installation/update date:	22/oct/2017
IDE name:					Arduino IDE
Flash Frequency:				40Mhz?
Upload Speed:				?115200?

### Description:
Installed arduino-esp32 yesterday on iMac OS 10.11.6 according Mac instructions on this github.
Downloaded latest Arduino 1.8.4
Selected ES32 Dev Module as board
Opened basic sketch
Compile results in this error (more details below)
cc1plus: internal compiler error: Illegal instruction: 4

What I tried so far:
- checkd on Google for cc1plus errors and possible caused but did not find anything
- checked "issues" on this github
- checked that directory arduino/hardware/espressif/hardware/esp32 is complete. Directory extensa-esp32-elf is present.
- checked that xcode-select is uptodate
- checked that other Arduino1.8.1 instaal for ESP8266 compiles OK, it does!
- looked at compile details to find a clue what is wrong. No clear indication for me.

So what could be the problem?

================error message details from arduino===========================

Arduino: 1.8.4 (Mac OS X), Board:"ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 921600, None"

/Applications/Arduino1.8.4.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino1.8.4.app/Contents/Java/hardware -hardware /Users/gerard/Library/Arduino15/packages -hardware /Users/gerard/Documents/Arduino/hardware -tools /Applications/Arduino1.8.4.app/Contents/Java/tools-builder -tools /Applications/Arduino1.8.4.app/Contents/Java/hardware/tools/avr -tools /Users/gerard/Library/Arduino15/packages -built-in-libraries /Applications/Arduino1.8.4.app/Contents/Java/libraries -libraries /Users/gerard/Documents/Arduino/libraries -fqbn=espressif:esp32:esp32:FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10804 -build-path /var/folders/5l/fjrz26dn5710yvf5hgzlcvsw0000gn/T/arduino_build_341798 -warnings=all -build-cache /var/folders/5l/fjrz26dn5710yvf5hgzlcvsw0000gn/T/arduino_cache_689403 -prefs=build.warn_data_percentage=75 -verbose /Users/gerard/Documents/Arduino/hardware/espressif/esp32/libraries/ESP32/examples/ChipID/GetChipID/GetChipID.ino
/Applications/Arduino1.8.4.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino1.8.4.app/Contents/Java/hardware -hardware /Users/gerard/Library/Arduino15/packages -hardware /Users/gerard/Documents/Arduino/hardware -tools /Applications/Arduino1.8.4.app/Contents/Java/tools-builder -tools /Applications/Arduino1.8.4.app/Contents/Java/hardware/tools/avr -tools /Users/gerard/Library/Arduino15/packages -built-in-libraries /Applications/Arduino1.8.4.app/Contents/Java/libraries -libraries /Users/gerard/Documents/Arduino/libraries -fqbn=espressif:esp32:esp32:FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10804 -build-path /var/folders/5l/fjrz26dn5710yvf5hgzlcvsw0000gn/T/arduino_build_341798 -warnings=all -build-cache /var/folders/5l/fjrz26dn5710yvf5hgzlcvsw0000gn/T/arduino_cache_689403 -prefs=build.warn_data_percentage=75 -verbose /Users/gerard/Documents/Arduino/hardware/espressif/esp32/libraries/ESP32/examples/ChipID/GetChipID/GetChipID.ino
Using board 'esp32' from platform in folder: /Users/gerard/Documents/Arduino/hardware/espressif/esp32
Using core 'esp32' from platform in folder: /Users/gerard/Documents/Arduino/hardware/espressif/esp32
Detecting libraries used...
"/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/config" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/app_trace" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bootloader_support" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bt" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/driver" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_adc_cal" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/ethernet" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/fatfs" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/heap" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/jsmn" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/log" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/mdns" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls_port" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/soc" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/sdmmc" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/spiffs" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/ulp" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/vfs" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/wear_levelling" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/xtensa-debug-module" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/console" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/coap" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/wpa_supplicant" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/expat" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/json" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/nghttp" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/lwip" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti  -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10804 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32  -DESP32 -DCORE_DEBUG_LEVEL=0 "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/cores/esp32" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/variants/esp32" "/var/folders/5l/fjrz26dn5710yvf5hgzlcvsw0000gn/T/arduino_build_341798/sketch/GetChipID.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/config" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/app_trace" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bootloader_support" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bt" "-I/Users/gerard/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/driver" "-I/Users/gerard/Documents/Arduino/h [... cut ...] ders/5l/fjrz26dn5710yvf5hgzlcvsw0000gn/T/arduino_build_341798/sketch/GetChipID.ino.cpp" -o "/var/folders/5l/fjrz26dn5710yvf5hgzlcvsw0000gn/T/arduino_build_341798/preproc/ctags_target_for_gcc_minus_e.cpp"
cc1plus: internal compiler error: Illegal instruction: 4
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
exit status 1
Fout bij het compileren voor board ESP32 Dev Module

me-no-dev, 2017-10-23:
huh... no clue to me either. I am using the same Arduino and have been compiling all day. Could it be an error downloading the toolchain or unarchiving it?

gerardwr, 2017-10-23:
huh..... that's what I thought!

I have been working with Arduino+ESP8266 since I did some beta-testing for iggr, so I expected no problems to work with the ESP32.

While experimenting I downloaded arduino-ESP32 on several occasions. so that's not likely to be the problem. Unarchiving is also unlikely, would be a first in many years.

But, to be sure I will try once more.

BTW, I am running Mac OS El Capitan 10.11.7. Are you on Siera of High Sierra?

me-no-dev, 2017-10-23:
Yes I run both Siera and High Sierra. Maybe the version matters for the toolchain? @igrr any comment?

gerardwr, 2017-10-23:
And I am running Java 1.6.0, do you use a newer version?

gerardwr, 2017-10-23:
So I installed arduino-esp32 all over again on my iMac, same result with the "cc1plus" error.

Just for fun I did the same install on a MacBook (same OS El Capitan 10.11.6). Result, NO ERROR ???!!

So it seems that there is something "weird" going on in the iMac.

The iMac problem is not yet solved, but the githubversion is OK! Apologies for the panic ;-)

me-no-dev, 2017-10-23:
please share what it was if you find it :)

closed 2017-10-23 by me-no-dev

gerardwr, 2017-10-25:
Re-installed xcode command line tools, but the cc1plus error remains ;-(
The search continues .......

```
cc1plus: internal compiler error: Illegal instruction: 4
```
Delete xcode command line tools
```
sudo rm -rf /Library/Developer/CommandLineTools
```

Install xcode command line tools
```
xcode-select --install
```

igrr, 2017-10-30:
Thanks for reporting this issue. We need to set macos_verison_min option when compiling on new hardware. It was done when the toolchain was built manually in the past, but the new CI build script doesn't set this option. So the toolchain ends up being incompatible with Macs which have older processors.

gerardwr, 2017-10-30:
@igrr 

Really !? That would be a great find, Ivan.

Just checked for fun:
The iMac with the cc1plus error has a T7300 Merom CPU.
The MacBook without the error has a P7550 Penryn CPU. 
Both system running the same os, the same Arduino 1.8.4 and the latest Arduino-esp32 github.

So what’s the way forward in setting the option? I assume this option has to be set in the toolchain-build and can’t be set locally in my iMac, right?

gerardwr, 2017-11-24:
@iggr : any news on changing the CI build script for "older processors"? Would be a shame if I had to scrap my 20" iMac ;-)

iggr, 2017-11-24:
Please refer your question to @igrr

gerardwr, 2017-11-24:
@iggr : changed to @igrr, my bad!

gerardwr, 2018-01-07:
Hi,

AFAICS this issue is not resolved yet. Is there any news re. solving this issue, please!

gerardwr, 2018-02-16:
This issue is still not resolved, but building your own XTENSA toolchain is a workaround. Reserve a few hours to get it going.
https://github.com/espressif/esp-idf/issues/1488