espressif/arduino-esp32 #765: Change STA

holema, 2017-10-23:
### Hardware:
Board:							Node 32S
Core Installation/update date:			23.10.2017
IDE name:						Platform.io
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
I tried to change the STA during the runtime with a Websockecket command. On the ESP8266 it was posible to use:

```cpp
	Serial.println(input);		
	WiFi.disconnect();
	WiFi.begin(daten["daten"][0].as<const char*>(), daten["daten"][1].as<const char*>());
```
To change the STA

but with the ESP32 after using the above code, I have no connection and the status Code 1, that the SSID is not available. 
I connected the ESP sucessfull to a router and changed the connection with the above commend. 

Then it disconnects from the network and gives the Status 1 (no SSID found).

copercini, 2017-10-27:
`Serial.println(daten["daten"][0].as<const char*>());` shows the correct SSID?

holema, 2017-10-27:
Yes, it returns the correct ssid.

holema, 2017-11-03:
It is working now correctly.
I did the WiFi.begin() command in the asyncWebSocket Routine which leads to the failure.
Now I set an flag and do the connectin in the main loop.

This works like a charme:

```cpp
Serial.printf("Connection status: %d\n", WiFi.status());
Serial.println(WiFi.status());
if(WiFi.status() == WL_CONNECTED){
     WiFi.disconnect();
     Serial.printf("Connection status: %d\n", WiFi.status());
     while (1) {
         if(WiFi.status() != WL_CONNECTED){
         break;
          }
    }
}
delay(100);

if (WiFi.status() != WL_CONNECTED) {
	Serial.printf("Connection status: %d\n", WiFi.status());
	WiFi.begin(ssid.c_str(), ssidPassword.c_str());
	WiFi.reconnect();
	WiFi.waitForConnectResult();
	Serial.printf("Connection status: %d\n", WiFi.status());
}
```

closed 2017-11-03 by holema