espressif/arduino-esp32 #766: Invalid library BLE

Testato, 2017-10-23:
Updated today to last master repo, on arduino IDE is there this warning
`Invalid library found in \SKETCH\hardware\espressif\esp32\libraries\BLE:`

me-no-dev, 2017-10-23:
run ```git submodule update --init --recursive``` in the esp32 arduino folder

Testato, 2017-10-23:
No git here :-)

nodtem66, 2017-10-24:
For platformIO:
I rewrite ***platformio.ini*** into
```ini
[env:your_target]
platform = https://github.com/platformio/platform-espressif32.git#feature/stage
framework = arduino
```
For arduino ide:
I suggested to download the latest version (**master** branch) of this git and replace in `esp32-arduino` installation folder

Curclamas, 2017-10-24:
I suggest to update the [guide](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md), since there is now a submodule, we have to clone recursively:
```git clone --recursive -j8 https://github.com/espressif/arduino-esp32.git```

aurelien35, 2017-11-23:
As Curclamas said, I also suggest to update the guide, which is really clear (even for beginners) but missing this step

hiphigi, 2018-01-07:
Even after trying Curclamas' suggestion I get the same error. Has anything changed? Seems that even the guide itself (Windows) is incomplete at this time.

closed 2019-08-15 by stale[bot]