espressif/arduino-esp32 #769: Upload Issue: Cannot run program "{tools.esptool.path}.exe": CreateProcess error=2, The system cannot find the file specified

HuanchengHu, 2017-10-24:
I am working on Infrared Sensor MLX 90614 on my ESP32. I used the example sketch and got successful compile. But when I tried to upload it shows error:

java.io.IOException: Cannot run program "{tools.esptool.path}.exe": CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:207)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:713)
	at processing.app.SketchController.exportApplet(SketchController.java:686)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2168)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 8 more
An error occurred while uploading the sketch

Can anyone help me? Thank you!

me-no-dev, 2017-10-24:
ughh Windows :( I have an idea ;) give me a few

HuanchengHu, 2017-10-24:
Thank you very much!

mihaly4, 2017-10-24:
I gues I am experiencing same problem on ArchLinux. Everything compiles OK, but upload failed.
Here is my log:
`python {tools.esptool.path}.py --chip esp32 --port /dev/ttyUSB1 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 /home/dev/Arduino/hardware/espressif/esp32/tools/partitions/boot_app0.bin 0x1000 /home/dev/Arduino/hardware/espressif/esp32/tools/sdk/bin/bootloader_qio.bin 0x10000 /tmp/arduino_build_730196/WiFiSmartConfig.ino.bin 0x8000 /tmp/arduino_build_730196/WiFiSmartConfig.ino.partitions.bin 
python: can't open file '{tools.esptool.path}.py': [Errno 2] No such file or directory
An error occurred while uploading the sketch`

pelomedusa, 2017-10-24:
Found any solution?

me-no-dev, 2017-10-24:
solution i will find :) can't find time to turn on the windows VM :) will do my best

Curclamas, 2017-10-24:
Same problem for me. I can mitigate it by `git checkout 55289a4`. Looks like the problem is induced by 4f55293f5901c4d060c1d58540988e0ac9739a3b and/or 84b8582acbb8ea76cd8b31400b7ff6edc98afc94

pelomedusa, 2017-10-24:
I'll check tonight and will give feddback. Thanks a lot

HuanchengHu, 2017-10-24:
git checkout command doesn't change anything on my device :(

Curclamas, 2017-10-24:
I forgot I had to also (after the checkout) delete the `/AppData/Local/Arduino15`, caution: all Arduino settings are lost (while the other folders are -of course- still preserved. That fixed it for me

HuanchengHu, 2017-10-24:
It works! Thank you very much!

me-no-dev, 2017-10-24:
Please try the latest commit

githubcdr, 2017-10-24:
working here

Curclamas, 2017-10-24:
@HuanchengHu @me-no-dev  After aa9111adfa7cdb52d1d6b3385d58ff1d8734c5a6 it is fixed for me.

jestebanes, 2017-10-24:
Now working again, thanks

me-no-dev, 2017-10-24:
great :) closing!

closed 2017-10-24 by me-no-dev