espressif/arduino-esp32 #770: Invalid Library found

Hasanshovon, 2017-10-24:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			24/10/2017
IDE name:							Arduino IDE 1.8.5
Flash Frequency:					80Mhz
Flash mode:	                                       QIO
Upload Speed:						115200

### Description:
my ESP32 dev board's library is not working .i have istalled library from (https://github.com/espressif/arduino-esp32/blob/master/README.md) Arduino core for ESP32 WiFi chip Build Status .and also followed instruction from "Steps to install Arduino ESP32 support on Windows" the link is(https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md) .but after completing above things when i simply upload blink code the compiler sent me the below error.......
what can i do now ???????????????? 

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(2, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(2, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(2, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}

### Debug Messages:
```
# [image]

Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

Curclamas, 2017-10-24:
It looks like it could be possibly be related to issue #769 and #766

me-no-dev, 2017-10-24:
Please try the latest commit

Curclamas, 2017-10-24:
@me-no-dev  @Hasanshovon The "red-error" (regarding esptool-path) is fixed by aa9111adfa7cdb52d1d6b3385d58ff1d8734c5a6 for me. The "Invalid Library found" could be related to #766.

me-no-dev, 2017-10-24:
@Curclamas that "Invalid Library found" is irrelevant :) check the path ;)

closed 2017-10-24 by me-no-dev