espressif/arduino-esp32 #772: OTA firmware

rel1ct, 2017-10-24:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			24/10/2017
IDE name:							Arduino IDE 1.8.5
Flash Frequency:					40Mhz
Upload Speed:						921600

### Description:
In comparison with esp8266, the firmware file is downloaded for a very long time.
I'm aware that the firmware size in esp32 is more than twice, but the load time is incommensurably larger (7 sec. for esp8266 and 45 sec. for esp32).

[esp32.pcapng](https://cloud.mail.ru/public/6coh/2ifQcdkGL)

### Sketch:
```cpp

#include "WiFi.h"
#include "WebServer.h"
#include "Update.h"
WebServer server(80);

char* ssid_ap = "ESP32";
char* pass_ap = "1234567890";
unsigned long time1,time2;

void setup(void) {
  Serial.begin(115200);
  WiFi.mode(WIFI_AP);
  WiFi.softAP(ssid_ap, pass_ap);
  updateOTA();
  server.begin();
}

void loop(void) {
  server.handleClient();
}

void updateOTA() {
  server.on("/update", HTTP_POST, [](){
    server.sendHeader("Connection", "close");
    server.send(200, "text/plain", (Update.hasError())?"FAIL":"OK");
    delay(500);
    ESP.restart();
  },[](){
    HTTPUpload& upload = server.upload();
    if(upload.status == UPLOAD_FILE_START){
      time1 = millis()/1000;
      uint32_t maxSketchSpace = 0x140000;
      if(!Update.begin(maxSketchSpace)){
        Update.printError(Serial);
      }
    } else if(upload.status == UPLOAD_FILE_WRITE){
      if(Update.write(upload.buf, upload.currentSize) != upload.currentSize){
        Update.printError(Serial);
      }
    } else if(upload.status == UPLOAD_FILE_END){
      if(Update.end(true)){
        time2 = millis()/1000; 
        Serial.printf("Update Success: %u B\nDownload time: %d sec.\nRebooting...\n", upload.totalSize, time2 - time1);
      } else {
        Update.printError(Serial);
      }
    }
    yield();
  });
}
```

###Port monitor:
```
(esp32)
Update Success: 501072 B
Download time: 45 sec.
Rebooting... 

(esp8266)
Update Success: 249072 B
Download time: 7 sec.
Rebooting..
```
[WebServer.h](https://github.com/bbx10/WebServer_tng)

me-no-dev, 2017-10-24:
Interesting capture :)
I can conclude a few things that I see:
1. After ACK the computer seem to wait for Window Update 120-160ms.
2. After the 7th second the browser actually catches up and start filling the buffer again and waits from Window Update to send more

I also notice slowdown when data is going into the ESP32 but have not looked that deep into it.
I doubt that it takes 160ms to write each packet to flash.

Some speed tests (not ideal conditions) without FS (Upload/Download) to my Async Web Server
ESP32 STA Upload: 4.10 Mbps
ESP32 STA Download: 9.02 Mbps
ESP32 AP Upload: 7.94 Mbps
ESP32 AP Download: 9.40 Mbps
ESP32 ETH Upload: 15.53 Mbps
ESP32 ETH Download: 23.12 Mbps
ESP8266 STA Upload: 9.65 Mbps
ESP8266 STA Download: 11.81 Mbps
ESP8266 AP Upload: 7.45 Mbps
ESP8266 AP Download: 9.27 Mbps

me-no-dev, 2017-10-24:
BTW just tested with my web server and here are the results (last column is millis):
```
Update Start: arduhal.bin, 11279
[D][Updater.cpp:123] begin(): OTA Partition: app1
Update Success: 518208B, 17894
Update Finished: OK, 17896

Update Start: arduhal.bin, 34996
[D][Updater.cpp:123] begin(): OTA Partition: app1
Update Success: 518208B, 41604
Update Finished: OK, 41607
```

rel1ct, 2017-10-25:
[8266.pcapng](https://cloud.mail.ru/public/6jat/oLBDAN1xV) for comparison
In esp32:
`7.110866 	192.168.4.1	192.168.4.2	TCP	54	[TCP ZeroWindow] 80 → 1229 [ACK] Seq=1 Ack=66057 Win=0 Len=0`
can not receive data ???

closed 2017-10-29 by rel1ct