espressif/arduino-esp32 #776: Make Serial1 and Serial2 more like Arduino

ayavilevich, 2017-10-25:
### Hardware:
Board:							ESP32 DevKitV1
Core Installation/update date:			20/oct/2017
IDE name:							Arduino IDE and VisualMicro
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
(AVR) Arduino creates Serial1 and Serial2 global instances of HardwareSerial if those are available. Arduino also defines HAVE_HWSERIAL1 and HAVE_HWSERIAL2 preprocessor definitions.
ESP32 Arduino only creates a Serial for UART0.

If no objection, please add those to the ESP32 Arduino as well to help with backward compatibility and reduce ifdefs.

Thanks.

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
#ifdef HAVE_HWSERIAL2
Serial2.begin();
#endif
}

void loop() {
}
```

stickbreaker, 2017-10-27:
By Default Serial1 is not unable. if you don't specify different pins other than (9,10) you will get a core dump.  GPIO9,10 are used by the Attached FLASH.
So, which pins do you think should always be assigned to Serial1? 

The AVR MEGA which has 4 predefined Serial has 54 I/O pins whereas the ESP32 only 26  so reserving any pin is more resource expensive.  I would rather force the user to make the decision.

(just my 2 cents).

Chuck.

ayavilevich, 2017-10-27:
@stickbreaker, you are right, but this is not a reason not to define Serial1 and HAVE_HWSERIAL1. The definition doesn't allocate the pins. The crash will only happen if you issue a begin() with default parameters and the SPI is on those pins.

I think the issue of uart1 sharing default pins with spi flash is a separate issue and I don't know why is was done this way. To mitigate this, I suggest a comment on the Serial1 property to warn users that there might be a conflict with spi flash and that a begin shoudl be called with custom pins where appropriate.

me-no-dev, 2017-10-30:
I do not get the point for ```HAVE_HWSERIAL1``` all ESP32 have 3 UARTS :)
Adding what you want is not a big deal (predefine Serial1 and Serial2)

ayavilevich, 2017-10-30:
It is about being able to support cross platform code without too many "#ifdef ESP32".
I know it is easy to define a Serial1 or 2 myself, but any existing project expects those to exists per Arduino convention.

closed 2019-08-15 by stale[bot]