espressif/arduino-esp32 #779: WPS : disconnected from station in loop

bhaessig83, 2017-10-26:
### Hardware:
Board:							Olimex ESP32-GATEWAY
Core Installation/update date:		26/10/2017
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					115200

### Description:
I've compiled the WPS.ino (of the WiFi example folder) and loaded into the controller and I get in loop the message "Disconnected from station, attempting reconnection" as long as the WPS is active on my router.

It seems I do not get the right status from WifiEvent

### Debug Messages:
In the monitor I see:
```
Starting WPS
Station Mode Started
Disconnected from station, attempting reconnection
Disconnected from station, attempting reconnection
Disconnected from station, attempting reconnection
Disconnected from station, attempting reconnection
Disconnected from station, attempting reconnection
Disconnected from station, attempting reconnection
Disconnected from station, attempting reconnection
...
```
Thank you for your help.
Bertrand

me-no-dev, 2017-10-30:
Please switch to ESP32 Dev Board in the boards menu and enable Log level :) it would say more about what events it get's and maybe give us a clue.

bhaessig83, 2017-10-30:
Hi,
I've selected the ESP32 dev module and activated the Core Debug Level "Info".
I get the following feedback in the monitor:

> Starting WPS
> Station Mode Started
> [W][WiFiGeneric.cpp:270] _eventCallback(): Reason: 201 - AUTH_FAIL
> Disconnected from station, attempting reconnection
> [W][WiFiGeneric.cpp:270] _eventCallback(): Reason: 201 - AUTH_FAIL
> Disconnected from station, attempting reconnection
> [W][WiFiGeneric.cpp:270] _eventCallback(): Reason: 201 - AUTH_FAIL
> Disconnected from station, attempting reconnection
> [W][WiFiGeneric.cpp:270] _eventCallback(): Reason: 201 - AUTH_FAIL
> Disconnected from station, attempting reconnection
> ...

I hope it could help us

me-no-dev, 2017-10-30:
this happens on boot or after WPS? Maybe old WiFi data is causing this? Can you add WiFi.disconnect(true) after WiFi.mode?

bhaessig83, 2017-10-30:
Before the "WiFi.disconnect(true)" change I had this behaviour:
- "Starting WPS" and "Station mode Started" ar displayed on boot.
- The WPS errors are comming 1-2s after I press the WPS button of the router.

Now, I do not have any error, but the login is blocked at the stage "Starting WPS"

The setup looks now like this
`void setup(){
  Serial.begin(115200);
  delay(10);

  Serial.println();

  WiFi.onEvent(WiFiEvent);
  WiFi.mode(WIFI_MODE_STA);

  WiFi.disconnect(true);

  Serial.println("Starting WPS");

  esp_wifi_wps_enable(&config);
  esp_wifi_wps_start(0);
}`

me-no-dev, 2017-10-30:
OK, then it's not old data, but rather an issue with WPS. Can you please check the issues over at espressif/esp-idf and if no such issue is filed recently, file one :)

me-no-dev, 2017-10-30:
Please provide the minimum sketch and model router ;)

bhaessig83, 2017-10-30:
OK, I just opened a topic as you proposed: https://github.com/espressif/esp-idf/issues/1188

closed 2019-08-15 by stale[bot]