espressif/arduino-esp32 #782: GPIO 34 & 35 always reads as LOW with Pull-up

hasmqtt, 2017-10-27:
### Hardware:
Board:							?ESP32 WROOM32?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Platform.io?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
GPIO read on pin 34 and 35 returns LOW even I have used internal pull-up. One of post on ESP32 I have read as silicon bug and handled using RTC. 

Appreciate help or pointers on how to resolve it.

### Sketch:
```cpp
#include <Arduino.h>

const int switchGPIOs[] = {34, 35, 32, 33, 25, 26}; //Connected to Switches
const int totalSwitches = sizeof(switchGPIOs)/sizeof(switchGPIOs[0]);
const int ssrGPIOs[] = {27, 14, 12, 13, 15, 2}; //Connected to LEDs

void initGPIOs() {
  for (int i = 0; i < totalSwitches; i++) {
    pinMode(switchGPIOs[i], INPUT_PULLUP);
    pinMode(ssrGPIOs[i], OUTPUT);
    digitalWrite(ssrGPIOs[i], HIGH);
  }
}

void readSwitches() {
  for (int i = 0; i < totalSwitches; i++) {
    int value = digitalRead(switchGPIOs[i]);
    Serial.printf("Value read is %d ==> %d\n", i, value);
    digitalWrite(ssrGPIOs[i], value);
  }
}

void setup() {
 Serial.begin(115000);
 initGPIOs();
}

void loop() {
readSwitches();
delay(30);
}
```

### Debug Messages:
None

MarkyAD, 2017-10-27:
Those pins (gpio34-39) dont have software pullup/down functions. Use an external pullup resistor, 10K works here.

hasmqtt, 2017-10-27:
Thanks for reply, then I will have to redesign my hardware :(

closed 2017-10-27 by hasmqtt

NickChungVietNam, 2018-06-05:
MarkyAD  Thank you.

TroyFletcher, 2019-11-26:
> Those pins (gpio34-39) dont have software pullup/down functions. Use an external pullup resistor, 10K works here.

THANK YOU! This was driving me crazy.

csobrinho, 2020-08-14:
Also drove me crazy for the last 2 days. Added a 10K and magic...
😖😖😖

malachib, 2023-02-22:
This saved me a lot of time, thank you!  For those extra picky people (like me) who want to see it in the official docs, [here it is](https://docs.espressif.com/projects/esp-idf/en/v5.0.1/esp32/api-reference/peripherals/gpio.html#_CPPv418gpio_set_pull_mode10gpio_num_t16gpio_pull_mode_t)

amandedhia, 2025-01-18:
Thanks for this! It is 3am and I was about to pull my hair out! Now I'm only pulling 10k