espressif/arduino-esp32 #784: onReceive in Wire.h?

cqin5, 2017-10-27:
### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
I noticed that in Wire.h there is no method similar to onReceive in the original Wire library, is it missing or is it not there by design?

If it is not there by design, what would be the alternative to it?

Thank you in advance.

stickbreaker, 2017-10-27:
Slave mode I2C of the ESP32 is markedly different than on the AVR chips.  The AVR `Wire.h` app callback method may not be possible.  It is still being worked on. 

Chuck.

pgrlopes, 2018-05-10:
Hello.

I'm having the same issue, is there any workaround for this other than including an alternate i2c library and going for lower level implementation?

copercini, 2018-09-21:
Close as duplicate of https://github.com/espressif/arduino-esp32/issues/118

closed 2018-09-21 by copercini