espressif/arduino-esp32 #785: OTA/UDP issue using eclipse, platformio works as expected. Why?

miknuc, 2017-10-27:
### Hardware:
Board:							ESP32 Dev Module (Doit)
Core Installation/update date:			21- OCT 2017 (fresh installation)
IDE name:						eclipse/ platformio
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
Hi,
I have an issue with OTA:
- when I use eclipse as IDE, I receive UDP errors and OTA does not work
- when I use the identical sketch in platformio, everything works as expected.

**Steps to reproduce the issue:**
**A. ecplise**
1. generate a new porject in ecplipse: git clone https://github.com/espressif/esp-idf-template.git
2.adapt eclipse project configuration according http://esp-idf.readthedocs.io/en/latest/get-started/eclipse-setup.html
3. copy https://github.com/espressif/arduino-esp32/tree/master/libraries/ArduinoOTA/examples/BasicOTA into eclipse project
4. correct ssid/WIFI PWD in BasixOTA.cpp
5. create Arduino esp32 components subdirectory
6. make menuconfig, adjust settings regarding usb port, usb_speed, ota partitions
7. make flash monitor and see what we get

Output:

Hard resetting... 
MONITOR 
ets Jun  8 2016 00:22:57^M 
^M 
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)^M 
ets Jun  8 2016 00:22:57^M 
^M 
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)^M 
configsip: 0, SPIWP:0xee^M 
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00^M 
mode:DIO, clock div:2^M 
load:0x3fff0010,len:4^M 
load:0x3fff0014,len:4480^M 
ho 0 tail 12 room 4^M 
load:0x40078000,len:0^M 
load:0x40078000,len:11756^M 
entry 0x40078ce4^M 
^[[0;33mW (72) rtc_clk: Possibly invalid CONFIG_ESP32_XTAL_FREQ setting (40MHz). Detected 40 MHz.^[[0m^M 
^[[0;32mI (141) cpu_start: Pro cpu up.^[[0m^M 
^[[0;32mI (141) cpu_start: Single core mode^[[0m^M 
^[[0;32mI (141) heap_init: Initializing. RAM available for dynamic allocation:^[[0m^M 
^[[0;32mI (144) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM^[[0m^M 
^[[0;32mI (150) heap_init: At 3FFB9810 len 000267F0 (153 KiB): DRAM^[[0m^M 
^[[0;32mI (157) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM^[[0m^M 
^[[0;32mI (163) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM^[[0m^M 
^[[0;32mI (169) heap_init: At 4008F9D4 len 0001062C (65 KiB): IRAM^[[0m^M 
^[[0;32mI (176) cpu_start: Pro cpu start user code^[[0m^M 
^[[0;32mI (44) cpu_start: Starting scheduler on PRO CPU.^[[0m^M 
ÿBooting^M 
I (59) wifi: wifi firmware version: 111e74d^M 
I (59) wifi: config NVS flash: enabled^M 
I (59) wifi: config nano formating: disabled^M 
^[[0;32mI (60) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE^[[0m^M 
^[[0;32mI (69) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE^[[0m^M 
I (84) wifi: Init dynamic tx buffer num: 32^M 
I (84) wifi: Init data frame dynamic rx buffer num: 32^M 
I (88) wifi: Init management frame dynamic rx buffer num: 32^M 
I (93) wifi: wifi driver task: 3ffc3190, prio:23, stack:4096^M 
I (99) wifi: Init static rx buffer num: 10^M 
I (102) wifi: Init dynamic rx buffer num: 32^M 
I (106) wifi: Init rx ampdu len mblock:7^M 
I (110) wifi: Init lldesc rx ampdu entry mblock:4^M 
I (114) wifi: wifi power manager task: 0x3ffc860c prio: 21 stack: 2560^M 
^[[0;32mI (141) phy: phy_version: 362.0, 61e8d92, Sep  8 2017, 18:48:11, 0, 0^[[0m^M 
I (142) wifi: mode : sta (XXXXXXXXXXXXXX)^M 
I (2551) wifi: n:11 0, o:1 0, ap:255 255, sta:11 0, prof:11^M 
I (3209) wifi: state: init -> auth (b0)^M 
I (3213) wifi: state: auth -> assoc (0)^M 
I (3218) wifi: state: assoc -> run (10)^M 
I (3240) wifi: connected with XXXXXXXX, channel 11^M 
^[[0;32mI (4217) event: sta ip: XXXXXXXX, mask: 255.255.255.0, gw: XXXXXXXX^[[0m^M 
**[E][WiFiUdp.cpp:58] begin(): could not set socket option: 109**^M 
**[E][ArduinoOTA.cpp:101] begin(): udp bind failed**^M 
Ready^M 
IP address: 192.168.1.52^M 
[E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9^M 
[E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9^M 
[E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9^M

OTA upload end in timeouts
I think the interesting lines are:

[E][WiFiUdp.cpp:58] begin(): could not set socket option: 109^M 
[E][ArduinoOTA.cpp:101] begin(): udp bind failed^M 

**B Platformio Test**
1. create new project
2. copy BAsixOTA.ino into project
3. adapt platformio.ini to USB connection
4. build

Result:

ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
Booting
Ready
IP address: XXXXXXXXX

OTA upload works as expected with platformio

Any explanation? I tried to compare the esp specific configuration in sdkconfig between eclipse and platformio but platformio seems to have a different approach to configure these settings.

me-no-dev, 2017-10-30:
you can find Arduino's sdkconfig in tools/sdk/sdkconfig :) use that for your project and you should be fine.

closed 2017-10-30 by me-no-dev

miknuc, 2017-10-30:
There might be a misunderstanding. I am not using the arduino ide but platformio. And platformio seems to have their own version of the esp32 sdk (somewhere in $HOME/.platformio/packages/framework-arduinoespressif32/), unfortunately without any sdkconfig templates.
I have some arduino directories from previous tests of the IDE, but I never tried them for esp32 and they do not contain any sdkconfig files.
A find in $HOME revealed only sdkconfig in $IDF_PATH subdirectories and in my esp-idf project directories.

me-no-dev, 2017-10-30:
in eclipse you use the github version of arduino and idf, right?
here is the file: https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/sdkconfig

miknuc, 2017-10-31:
Ok, I think I understood your suggestion finally :)
It works, thank you very much!

May I suggest to add that hint to replace the template sdkconfig with the arduino version in 
https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md
e.g.

mkdir -p components && \
cd components && \
git clone https://github.com/espressif/arduino-esp32.git arduino && \
cd arduino && \
git submodule update --init --recursive && \
cd ../.. && \
**cp components/arduino/tools/sdk/sdkconfig .**
make menuconfig

to help others that might run into similar issues?